AgOpenGPS-Official/AgOpenGPS #2: The sketch

aortner, 2016-10-29:
// Arduino Cerea Boom Control
// (C) 2016 by Andreas Ortner - office@gh-ortner.com
// 

String cmd;
char nextChar;
String teilbreite;
int anztb ;
int i;

void setup() {
  Serial.begin(9600);
  pinMode(2, OUTPUT);   //boom 1 Pin 2
  pinMode(3, OUTPUT);
  pinMode(4, OUTPUT);
  pinMode(5, OUTPUT);
  pinMode(6, OUTPUT);
  pinMode(7, OUTPUT);
  pinMode(8, OUTPUT);  // boom 7
  pinMode(9, OUTPUT);  // reley8 pin 9 hidraulic command

  digitalWrite(2, HIGH);  // all Ports  0 = HIGH 
  digitalWrite(3, HIGH);
  digitalWrite(4, HIGH);
  digitalWrite(5, HIGH);
  digitalWrite(6, HIGH);
  digitalWrite(7, HIGH);
  digitalWrite(8, HIGH);
  digitalWrite(9, HIGH);
}

void loop()
{
  cmd = "";   // String clean
  ser();      // read serial
  
  if (cmd.startsWith("@CEREA;")) {    // if  Cerea start with boom control
    cerea();
  }
  if (cmd.startsWith("@HIDRAU")) {    //if hidrau start with hidrau
    hidrau();                         // if you dont need it erase
  }
  
  
}

void cerea()   //String 
{
   
  cmd.remove(0, 7); // @Cerea; remove

  int first = cmd.indexOf(';');   // find first ; 
  int second = cmd.indexOf(';', first + 1 );  // find second ; suchen
  cmd.remove(0, second + 1); // remove gps speed and minus 1 

  int ende = cmd.indexOf('END'); //find  "end"
  anztb = (ende - 2) / 2; // how many booms

  teilbreite = cmd.substring(0, ende - 3); //write booms to new string

 

  int b = 1;
  for (i = 0; i < anztb * 2 ; i = i + 2) { //switch booms
    b = b + 1;
    if (teilbreite.substring(i, i + 1) == "1")  digitalWrite(b, LOW);       //boom on
    if (teilbreite.substring(i, i + 1) == "0" ) digitalWrite(b, HIGH);      //boom off
  }

  for (i = anztb + 1 ; i < 8; i++) { //not used booms off
    digitalWrite(i + 1, HIGH);
  }
}

void hidrau()
{
  int ende = cmd.indexOf('EN'); //serch "end"
  cmd.remove(0, ende - 3); // remove rest
  
  
  if (cmd.substring(0, 1) == "1") {
   digitalWrite(9, LOW);         //hyro on  
  }
  if (cmd.substring(0, 1) == "0") {
    digitalWrite(9, HIGH);       //hyro off
  }
}

void ser()   // read serial
{
  do {
    if (Serial.available() > 0)
    {
      nextChar = Serial.read();
      if (nextChar >= 32) {
        cmd += nextChar;
      }
    }
  } while (nextChar != 10);  //wait for 

}

farmerbriantee, 2016-10-30:
Cool, thanks.

The one i use is a bit shorter and uses Port calls directly. In AgOpenGPS it OR's or AND's the individual bits and then sends just one byte to Arduino. All the relays turn on or off at exactly the same time.

int inByte = 0;         // incoming serial byte
  
void setup() {
    //Pins 8 thru 12 as output, 8 is LSB
    DDRB = B00011111;

    //PORTB turn off 8 thru 12
    PORTB = B00000000;
    
  Serial.begin(9600);
  while (!Serial) {}

  establishContact();  // send a byte to establish contact until receiver responds
}

void loop() {
  // if we get a valid byte, read analog ins:
  if (Serial.available() > 0)
  {
    // get incoming byte:
    inByte = Serial.read();
    Serial.print(inByte);

    PORTB = inByte;
 
  }
  
}

void establishContact() {
  while (Serial.available() <= 0) {
    Serial.print("X");   // send an initial string
    delay(500);
  }
}

aortner, 2016-10-30:
Hi Brian! I did see your new sketch to late.

At cerea they use another way.  The gps software sends a string to the arduino.

each command beginns with @ 
then the use strings to assign the commands to different functions.

@hidraulic;1 or 0;End   is for a hyraulic command to lift off the machine
@cera;gpsspeed;0 or 1;0 or 1; ..........;End     is for controlling the sections.  

maybe they use more commands in the future comming from the machine.

it is also possible the send commands from the arduino to the gpssoftware.

i have taken the allsprayer on/off signal from the sprayer to send it to gps software.
the software turns on/off the mark behind the trekkker. this is very usefull when driving a little outside the field. 

i did not look in you source code.  witch programming language is it?
what hardware are you using?

at the moment i am part of the german user group of cerea. cerea is a spanish software and the
support is terrible - but it is very cheap!

best regards andreas

office@gh-ortner.com
+43 660 7669333 (whatsapp)

farmerbriantee, 2016-10-30:
I'm using C# and the application runs in windows, but it will also run in Linux thru wine. It will run in any windows 7 to 10. Never tried in XP.

I've watched some Youtube videos on Cerea. It is quite good, but as you say support would be difficult.

closed 2018-04-05 by aortner