AgOpenGPS-Official/AgOpenGPS #7: TCP Input

aortner, 2017-03-18:
Hi! 

I am asking my question again. I had see that you have done something with tcp/ip.

Is it possible to get input from my reach unit?    192.168.2.15:9999

best regards Andreas

farmerbriantee, 2017-03-20:
Are you able to edit source code? Do you have visual studio 2015 or 2017 installed?

aortner, 2017-03-20:
I can edit source / but i dont like it ;-)

btw i can stream nmea data to anywere for testing using strsvr.
if you can give me ipadress and port i can do it so.

Andreas

farmerbriantee, 2017-03-20:
Search for the line " StartTCPServer();" in FormGPS.cs. 

Uncomment it so the TCP server actually starts when starting AgOpen. Rebuild project.

The IP address you already know, its the ip address of the machine you are running AgOpen on.

The port number is 7777.  In the function StartTCPServer, the line 

const int nPortListen = 7777;

sets the port number of the server. If you need to change that, no problem. 

In your client, put the ip address of the AgOpenGPS, and port 7777.

It should work!

lackolaxton, 2017-03-20:
Brian,
Would you like me to look at adding a GUI for the TCP/UDP server(s)? We can then add functionality to change the above settings while using the application. Depends if you deem it necessary and if so a little brief on what it neeeds to support would be great :)

Cheers
Tom

aortner, 2017-03-20:
Hi Brian,Hi Tom!

I did try your suggestions with sharp develop - i have no visual studio.

i was able to build the app and tcp input works - great.
i did loose some graphics...

if you could make a switch for this in your app - it would be great!

farmerbriantee, 2017-03-22:
It has to run when starting AgOpen, but would only require a RESTART then. i best add it.

farmerbriantee, 2017-03-22:
oh, but i've been feverishly working on fixing the heading when the gps signal isn't the best. Gonna take a lot to fix, its a super priority.

closed 2018-04-05 by aortner