AgOpenGPS-Official/AgOpenGPS #11: Issue with reconnection to serial port after serial device power cycle

svenn71, 2017-09-04:
When I stop and start my tractor during field work, the power to the device on the serial port (GPS) is lost while the tablet is still alive due to its internal battery. AgOpenGPS keeps the serial port open, but somehow does not pick up the data on the serial port until a disconnect followed by a reconnect is performed. I struggled a while until I discovered this procedure. Now that I know, it is not a big issue, but maybe the application could perform the disconnect and reconnect automatically a couple of times when it discovers that no data is coming in on the serial port.

The issue is probably related to me using an externally powered OTG hub for the serial-to-USB converter and windows 10 actually re-enumerates the USB device when the power comes back, while AgOpenGPS listens to a stale handler, or something like that.

farmerbriantee, 2017-10-02:
For sure! That is very narrow so yes it would only show zero. I will add that to the source. Thanks!

farmerbriantee, 2017-10-02:
Right comment, wrong question! 

Serial ports are a real tough one in windows and C#.  Also the serial ports are asynchronous - they don't wait for the main program and run in separate threads. Once that connection is lost, it needs to be manually reconnected. I tried really hard to get it to reconnect, but was unable. That being said though, a good installation makes sure power is provided to everything all the time - then no reconnection is required and makes it all very reliable. If it disconnects, it generally means there is a problem with the power and or installation. Ive seeded thousands of acres with the sectional relay part reading an RTK based GPS system, never lost communication. 

Maybe one day i will figure out the reconnection thing. Its a tough one

svenn71, 2017-10-02:
Hi,
On most of the vehicles I have, there is no easy connection to permanent 12v. Even when connecting directly to the battery the start engine may pull the voltage low enough for some of the devices on aux power to fall out. I am thinking about using a smaller secondary battery as a buffer.

I have found some indications on the net that some rs232-usb cables confuse their driver if they receive rs232 while being brought up by a booting host system. In my configuration, the SF300 antenna starts spitting out data almost immediately after power on while the Acer Switch one 10 take a while to respond on my button press to kick it alive. This causes the need to shut down AOG, unplug the usb-rs232 cable and then reconnect and start up AOG again. Solution is to keep the SF300 unpowered until the Acer is up running and then power the antenna.

A second setup that I have with an 8" windows tablet does not see this particluar problem with the need of a hard shutdown and disconnect when I start the vehicle after lunch break. There are tons of setup regarding sleep/hibernate when in power mode and battery mode on the terminals that may be different.

The deep failure to restart GPS capture seems to be very system setup dependent. I'll start with a simple power switch on my gps power.

farmerbriantee, 2017-10-02:
Yes it really is dependent on setup and hardware. I don't know how to make it work in all situations.

Uploaded new version on github today, yay!

Follow along here for discussion if you want...
http://www.thecombineforum.com/forums/31-technology/296233-diy-autosteer-agopengps.html#post2993233

closed 2018-04-05 by farmerbriantee

mirh, 2022-04-29:
Some ideas
https://stackoverflow.com/questions/11890940/reconnecting-to-com-after-lost-connection/#11899051
https://stackoverflow.com/questions/13408476/detecting-when-a-serialport-gets-disconnected
https://stackoverflow.com/questions/4199083/detect-serial-port-insertion-removal
https://stackoverflow.com/questions/37777626/how-can-my-application-recover-when-a-usb-emulated-serial-port-vanishes-and-reap
and some samples
https://github.com/fivepmtechnology/SharpGPS/blob/master/src/SharpGps/SerialPort.cs
https://github.com/dotMorten/NmeaParser/blob/main/src/NmeaParser/SerialPortDevice.Desktop.cs
https://github.com/jperson2000/GPS.Net/blob/master/Devices/SerialPort.cs
https://github.com/DotSpatial/DotSpatial/blob/master/Source/Core/DotSpatial.Positioning/SerialDevice.cs
(putting aside that the F9P has so many advanced settings that it seems hard none could make a difference here)