AgOpenGPS-Official/AgOpenGPS #13: Adding an aim for manual steering

svenn71, 2017-09-28:
This patch will add an aim in the form of a diamond on a rod a few meters in front of the triangle that represents the vehicle. I find it easier to drive along a line like an AB or contour guide line  when such a feature is included. I have not found the optimum geometry for the feature, but it is more a proof-of-concept than a finished feature. Taking hard turns is outside the scope of the aim.

```
 SourceCode/GPS/Classes/CVehicle.cs | 22 +++++++++++++++++++++-
 1 file changed, 21 insertions(+), 1 deletion(-)

diff --git a/SourceCode/GPS/Classes/CVehicle.cs b/SourceCode/GPS/Classes/CVehicle.cs
index 0e9828b..0c47738 100644
--- a/SourceCode/GPS/Classes/CVehicle.cs
+++ b/SourceCode/GPS/Classes/CVehicle.cs
@@ -214,7 +214,27 @@ public void DrawVehicle()
             gl.Vertex(1.8, -antennaPivot, 0.0);
             gl.End();
 
-           //draw the area side marker
+            gl.Begin(OpenGL.GL_LINES);
+            gl.Vertex(0, 0, -0.2);
+            gl.Vertex(0, antennaPivot + wheelbase + 1, -0.2);
+            gl.End();
+
+            gl.Begin(OpenGL.GL_LINE_LOOP);
+            for (int ii = 0; ii < 4; ii++)
+            {
+                double r = 1.0;
+                //get the current angle 
+                double theta = 2.0f * 3.1415926f * (double)(ii) / (double)(4);
+                //calculate the x component 
+                double x = r * Math.Cos(theta);
+                //calculate the y component 
+                double y = r * Math.Sin(theta);
+                //output vertex 
+                gl.Vertex(x + 0, y + antennaPivot + wheelbase + 1, -0.2);
+            }
+            gl.End();
+
+            //draw the area side marker
             gl.Color(0.95f, 0.90f, 0.0f);
             gl.PointSize(4.0f);
             gl.Begin(OpenGL.GL_POINTS);
```

farmerbriantee, 2017-10-02:
Cool. You can draw it out manually and save a lot of calculations like this....

           gl.LineWidth(3);
            gl.Color(0.9, 0.5, 0.30);
            gl.Begin(OpenGL.GL_LINE_STRIP);
            //for (int ii = 0; ii < 4; ii++)
            {
                gl.Vertex(1, -antennaPivot + wheelbase + 5, 0.0);
                gl.Vertex(0, -antennaPivot + wheelbase + 8, 0.0);
                gl.Vertex(-1, -antennaPivot + wheelbase + 5, 0.0);
            }
            gl.End();

svenn71, 2017-10-02:
Keeping calculations to a minimum seems to be quite crucial. The windows tablet that I use in one setup is draining its battery even if I supply 2.1A power. If I shut down AOG when taking a longer break filling up the sprayer, I give the system a bit of slack to recharge.

Maybe a 'pause' switch in AOG to stop the timing loop temporarily would be an option to shutting down AOG. Not that it is a problem to do that once I found out what to do to stretch my battery life time.

closed 2018-04-05 by farmerbriantee