AgOpenGPS-Official/AgOpenGPS #24: Antenna Offset and roll - vice versa

aortner, 2018-04-07:
antenna offset and roll are vice versa now.

WTalen, 2018-04-07:
As to say looking at back of tractor, tilting the right side up is negative? And an antenna on the right side of the tractor would be negative now?
[image]

farmerbriantee, 2018-04-08:
I never changed anything. Roll polarity is the same for a long time now.

aortner, 2018-04-08:
Roll - maybe my mistake in my setup of my gps..

but antenna offset is the wrong way - if i rerun agopengps with 1 m - its to the right - but it should be to the left...

[image]

farmerbriantee, 2018-04-08:
If its in the wrong plaMAce, go the other direction with the setting. I know that's really obvious, but I'll try to standardize this stuff to follow the plane standard

aortner, 2018-04-08:
yes i did change setting the other direction..
-took my some time to find out was was wrong...

closed 2018-04-08 by aortner