AgOpenGPS-Official/AgOpenGPS #43: Load border from kml

svenn71, 2018-06-17:
The KML import didn't work on my kml file, so I dug into the source to find the spot where the data is loaded and added some sauce.

``` 
SourceCode/GPS/Forms/FormBoundary.cs | 4 +++-
 1 file changed, 3 insertions(+), 1 deletion(-)

diff --git a/SourceCode/GPS/Forms/FormBoundary.cs b/SourceCode/GPS/Forms/FormBoundary.cs
index 295de1d..85ce168 100644
--- a/SourceCode/GPS/Forms/FormBoundary.cs
+++ b/SourceCode/GPS/Forms/FormBoundary.cs
@@ -113,7 +113,8 @@ private void btnLoadBoundaryFromGE_Click(object sender, EventArgs e)
                         {
                             line = reader.ReadLine();
                             line = line.Trim();
-                            string[] numberSets = line.Split(' ');
+                            char[] delimiterChars = { ' ', '\t', '\r', '\n' };
+                            string[] numberSets = line.Split(delimiterChars);
                             done = true;
 
                             //at least 3 points
@@ -124,6 +125,7 @@ private void btnLoadBoundaryFromGE_Click(object sender, EventArgs e)
                                 foreach (var item in numberSets)
                                 {
                                     string[] fix = item.Split(',');
+                                    if (fix.Length < 2) continue;
                                     double.TryParse(fix[0], NumberStyles.Float, CultureInfo.InvariantCulture, out lonK);
                                     double.TryParse(fix[1], NumberStyles.Float, CultureInfo.InvariantCulture, out latK);
                                     DecDeg2UTM(latK, lonK);
```

farmerbriantee, 2018-06-30:
I can't tell what was added?

svenn71, 2018-06-30:
First, I added more delimiter chars by 
char[] delimiterChars = { ' ', '\t', '\r', '\n' };
string[] numberSets = line.Split(delimiterChars);
as my gps give me kml data separated by \n

Then I made sure that the fix array contain at least two numbers by
if (fix.Length < 2) continue;
because line.split will insert extra empty items in numberSets on stray delimiters.

janide, 2018-08-20:
... is it possible to load other formats than KML?

svenn71, 2018-08-20:
I think AoG work in GPS coordinate system only.
you can use qgis to convert from local coordinates to gps and export as kml.

farmerbriantee, 2018-08-24:
I added your code. Thank you so much!

closed 2018-08-24 by farmerbriantee