AgOpenGPS-Official/AgOpenGPS #44: Offsets in Contour mode

WTalen, 2018-06-28:
When driving in contour mode if I use an offset, such as a disc mower 3.28m to the right, the guidance line is centered over the towed implement instead of the tractor.

farmerbriantee, 2018-07-20:
That's a much bigger problem then you might think. I will also have to save direction, and know which direction it went making the path, and also determine the direction of the path you want to drive on. One of those things that got "missed" because its really hard to do.

apmartin00, 2018-07-20:
to add complication, our mower can be either side of tractor as we mow following seeder lines, needs either steer angle sensor on mower pivot or gps antennae on centre of mower depending on how AOG is to be setup.