AgOpenGPS-Official/AgOpenGPS #48: AOG crashes on startup, perhaps LineUpManualBtns()?

Hb-, 2018-09-23:
AOG crashes on start up. 

The Windows event is:
> Application: AgOpenGPS.exe
> Framework Version: v4.0.30319
> Description: The process was terminated due to an unhandled exception.
> Exception Info: System.NullReferenceException
>    at AgOpenGPS.FormGPS.LineUpManualBtns()
>    at AgOpenGPS.FormGPS.FormGPS_Resize(System.Object, System.EventArgs)
>    at System.Windows.Forms.Control.OnResize(System.EventArgs)
>    at System.Windows.Forms.Form.OnResize(System.EventArgs)
>    at System.Windows.Forms.Control.OnSizeChanged(System.EventArgs)
>    at System.Windows.Forms.Control.UpdateBounds(Int32, Int32, Int32, Int32, Int32, Int32)
>    at System.Windows.Forms.Control.UpdateBounds(Int32, Int32, Int32, Int32)
>    at System.Windows.Forms.Control.SetBoundsCore(Int32, Int32, Int32, Int32, System.Windows.Forms.BoundsSpecified)
>    at System.Windows.Forms.Form.SetBoundsCore(Int32, Int32, Int32, Int32, System.Windows.Forms.BoundsSpecified)
>    at System.Windows.Forms.Control.ScaleControl(System.Drawing.SizeF, System.Windows.Forms.BoundsSpecified)
>    at System.Windows.Forms.ScrollableControl.ScaleControl(System.Drawing.SizeF, System.Windows.Forms.BoundsSpecified)
>    at System.Windows.Forms.Form.ScaleControl(System.Drawing.SizeF, System.Windows.Forms.BoundsSpecified)
>    at System.Windows.Forms.Control.ScaleControl(System.Drawing.SizeF, System.Drawing.SizeF, System.Windows.Forms.Control)
>    at System.Windows.Forms.ContainerControl.Scale(System.Drawing.SizeF, System.Drawing.SizeF, System.Windows.Forms.Control)
>    at System.Windows.Forms.ContainerControl.PerformAutoScale(Boolean, Boolean)
>    at System.Windows.Forms.ContainerControl.PerformNeededAutoScaleOnLayout()
>    at System.Windows.Forms.ContainerControl.OnLayoutResuming(Boolean)
>    at System.Windows.Forms.Control.ResumeLayout(Boolean)
>    at AgOpenGPS.FormGPS.InitializeComponent()
>    at AgOpenGPS.FormGPS..ctor()
>    at AgOpenGPS.Program.Main()

The "Problem signature " is 
> Problemsignatur:
>   Problemereignisname:	CLR20r3
>   Problemsignatur 01:	AgOpenGPS.exe
>   Problemsignatur 02:	2.7.0.0
>   Problemsignatur 03:	5b920428
>   Problemsignatur 04:	AgOpenGPS
>   Problemsignatur 05:	2.7.0.0
>   Problemsignatur 06:	5b920428
>   Problemsignatur 07:	16d
>   Problemsignatur 08:	8c
>   Problemsignatur 09:	System.NullReferenceException
>   Betriebsystemversion:	6.1.7601.2.1.0.256.48
>   Gebietsschema-ID:	1031
>   Zusatzinformation 1:	0a9e
>   Zusatzinformation 2:	0a9e372d3b4ad19135b953a78882e789
>   Zusatzinformation 3:	0a9e
>   Zusatzinformation 4:	0a9e372d3b4ad19135b953a78882e789

The AgOpenGPS.exe.CLRLoad00.log is:
2752,2623.344,CLR Loading log for C:\Users\hb\Desktop\AgOpenGPS-master\AgOpenGPS-master\Application\AgOpenGPS.exe
2752,2623.344,Log started at 22:20:04 on 23.09.18
2752,2623.344,-----------------------------------
2752,2623.344,FunctionCall: _CorExeMain
2752,2623.344,FunctionCall: ClrCreateInstance, Clsid: {2EBCD49A-1B47-4A61-B13A-4A03701E594B}, Iid: {E2190695-77B2-492E-8E14-C4B3A7FDD593}
2752,2623.344,MethodCall: ICLRMetaHostPolicy::GetRequestedRuntime.  Version: (null),  Metahost Policy Flags: 0x1168, Binary: (null), Iid: {BD39D1D2-BA2F-486A-89B0-B4B0CB466891}
2752,2623.344,Installed Runtime: v2.0.50727.  VERSION_ARCHITECTURE: 0
2752,2623.344,Installed Runtime: v4.0.30319.  VERSION_ARCHITECTURE: 0
2752,2623.344,Input values for ComputeVersionString follow this line
2752,2623.344,-----------------------------------
2752,2623.344,Default Application Name: C:\Users\hb\Desktop\AgOpenGPS-master\AgOpenGPS-master\Application\AgOpenGPS.exe
2752,2623.344,IsLegacyBind is: 0
2752,2623.344,IsCapped is 0
2752,2623.344,SkuCheckFlags are 0
2752,2623.344,ShouldEmulateExeLaunch is 0
2752,2623.344,LegacyBindRequired is 0
2752,2623.344,-----------------------------------
2752,2623.344,Parsing config file: C:\Users\hb\Desktop\AgOpenGPS-master\AgOpenGPS-master\Application\AgOpenGPS.exe.config
2752,2623.344,Config File (Open). Result:00000000
2752,2623.344,Config File (Read). Result:00000000
2752,2623.344,Found config file: C:\Users\hb\Desktop\AgOpenGPS-master\AgOpenGPS-master\Application\AgOpenGPS.exe.config
2752,2623.344,UseLegacyV2RuntimeActivationPolicy is set to 0
2752,2623.344,Config file includes SupportedRuntime entry.  Version: v4.0, SKU: .NETFramework,Version=v4.0
2752,2623.344,Found a supportedRuntime tag in the config file
2752,2623.344,Using supportedRuntime: v4.0.30319
2752,2623.344,Decided on runtime: v4.0.30319
2752,2623.406,MethodCall: ICLRRuntimeInfo::GetProcAddress.  Name: _CorExeMain
2752,2623.406,Runtime has been loaded.  Version: v4.0.30319
2752,2623.406,MethodCall: ICLRRuntimeInfo::GetVersionString.  Version: v4.0.30319
2752,2623.406,MethodCall: ICLRRuntimeInfo::GetRuntimeDirectory.  Version: v4.0.30319
2752,2623.406,Config File (Open). Result:00000000
2752,2623.422,Config File (Open). Result:00000000
2752,2623.422,MethodCall: ICLRRuntimeInfo::GetInterface.  Clsid: {F7721072-BF57-476D-89F8-A7625D27683A}, Iid: {07C4E752-3CBA-4A07-9943-B5F206382178}
2752,2623.422,MethodCall: ICLRRuntimeInfo::GetRuntimeDirectory.  Version: v4.0.30319
2448,2623.422,FunctionCall: OnShimDllMainCalled.  Reason: 2
2448,2623.422,FunctionCall: RealDllMain.  Reason: 2
2440,2623.422,FunctionCall: OnShimDllMainCalled.  Reason: 2
2440,2623.422,FunctionCall: RealDllMain.  Reason: 2
2440,2623.422,MethodCall: ICLRRuntimeInfo::GetVersionString.  Version: v4.0.30319
2752,2623.422,MethodCall: ICLRRuntimeInfo::GetVersionString.  Version: v4.0.30319
2752,2623.437,MethodCall: ICLRRuntimeInfo::LoadLibrary.  Name: clrjit.dll.  Version: v4.0.30319
4028,2623.453,FunctionCall: OnShimDllMainCalled.  Reason: 2
4028,2623.453,FunctionCall: RealDllMain.  Reason: 2
2752,2623.468,MethodCall: ICLRRuntimeInfo::LoadLibrary.  Name: nlssorting.dll.  Version: v4.0.30319
1008,2623.468,FunctionCall: OnShimDllMainCalled.  Reason: 2
1008,2623.468,FunctionCall: RealDllMain.  Reason: 2
4448,2623.468,FunctionCall: OnShimDllMainCalled.  Reason: 2
4448,2623.468,FunctionCall: RealDllMain.  Reason: 2
2752,2623.515,Config File (Open). Result:00000000
2768,2623.578,FunctionCall: OnShimDllMainCalled.  Reason: 2
2768,2623.578,FunctionCall: RealDllMain.  Reason: 2
2752,2623.687,Config File (Open). Result:00000000
2752,2623.983,Config File (Open). Result:00000000
2752,2623.983,MethodCall: ICLRRuntimeInfo::GetVersionString.  Version: v4.0.30319
1008,3162.795,FunctionCall: RealDllMain.  Reason: 3
1008,3162.795,FunctionCall: OnShimDllMainCalled.  Reason: 3
4284,3162.795,FunctionCall: OnShimDllMainCalled.  Reason: 2
4284,3162.795,FunctionCall: RealDllMain.  Reason: 2
1444,3162.795,FunctionCall: OnShimDllMainCalled.  Reason: 2
1444,3162.795,FunctionCall: RealDllMain.  Reason: 2

Hb-, 2018-09-24:
It looks like a bad .NET installation on this Windows 7 system causes the bug. A successful launch on a XP system gives the following CLRLoad log:

2076,4400.838,CLR Loading log for C:\Users\Hb\Desktop\AgOpenGPS-master\Application\AgOpenGPS.exe
2076,4400.838,Log started at 07:42:01 on 24.09.2018
2076,4400.838,-----------------------------------
2076,4400.838,LegacyFunctionCall: _CorExeMain
2076,4400.838,FunctionCall: ClrCreateInstance, Clsid: {2EBCD49A-1B47-4A61-B13A-4A03701E594B}, Iid: {E2190695-77B2-492E-8E14-C4B3A7FDD593}
2076,4400.838,MethodCall: ICLRMetaHostPolicy::GetRequestedRuntime.  Version: (null),  Metahost Policy Flags: 360, Binary: (null), Iid: {BD39D1D2-BA2F-486A-89B0-B4B0CB466891}
2076,4400.838,Installed Runtime: v2.0.50727.  VERSION_ARCHITECTURE: 0
2076,4400.838,Installed Runtime: v4.0.30319.  VERSION_ARCHITECTURE: 0
2076,4400.838,Input values for ComputeVersionString follow this line
2076,4400.838,-----------------------------------
2076,4400.838,Default Application Name: C:\Users\Hb\Desktop\AgOpenGPS-master\Application\AgOpenGPS.exe
2076,4400.838,IsLegacyBind is: 0
2076,4400.838,IsCapped is 0
2076,4400.838,SkuCheckFlags are 0
2076,4400.838,ShouldEmulateExeLaunch is 0
2076,4400.838,LegacyBindRequired is 0
2076,4400.838,-----------------------------------
2076,4400.848,Parsing config file: C:\Users\Hb\Desktop\AgOpenGPS-master\Application\AgOpenGPS.exe.config
2076,4400.848,Found config file: C:\Users\Hb\Desktop\AgOpenGPS-master\Application\AgOpenGPS.exe.config
2076,4400.848,UseLegacyV2RuntimeActivationPolicy is set to 0
2076,4400.848,Config file includes SupportedRuntime entry.  Version: v4.0, SKU: .NETFramework,Version=v4.0
2076,4400.848,Found a supportedRuntime tag in the config file
2076,4400.848,Using supportedRuntime: v4.0.30319
2076,4400.848,Decided on runtime: v4.0.30319
2076,4400.878,MethodCall: ICLRRuntimeInfo::GetProcAddress.  Name: _CorExeMain
2076,4400.878,Runtime has been loaded.  Version: v4.0.30319
2076,4400.878,MethodCall: ICLRRuntimeInfo::GetInterface.  Clsid: {F7721072-BF57-476D-89F8-A7625D27683A}, Iid: {07C4E752-3CBA-4A07-9943-B5F206382178}
2076,4400.878,MethodCall: ICLRRuntimeInfo::GetVersionString.  Version: v4.0.30319
2076,4400.878,MethodCall: ICLRRuntimeInfo::GetRuntimeDirectory.  Version: v4.0.30319
2076,4400.878,MethodCall: ICLRRuntimeInfo::GetVersionString.  Version: v4.0.30319
2276,4400.888,MethodCall: ICLRRuntimeInfo::GetVersionString.  Version: v4.0.30319
2076,4400.888,MethodCall: ICLRRuntimeInfo::LoadLibrary.  Name: culture.dll.  Version: v4.0.30319
2076,4400.888,MethodCall: ICLRRuntimeInfo::LoadLibrary.  Name: nlssorting.dll.  Version: v4.0.30319
2076,4400.888,MethodCall: ICLRRuntimeInfo::LoadLibrary.  Name: clrjit.dll.  Version: v4.0.30319
2076,4417.061,MethodCall: ICLRRuntimeInfo::GetInterface.  Clsid: {F7721072-BF57-476D-89F8-A7625D27683A}, Iid: {07C4E752-3CBA-4A07-9943-B5F206382178}

farmerbriantee, 2018-11-06:
Tried it on a few computers, seems to work fine. Still a problem?

closed 2018-12-10 by farmerbriantee

mirh, 2020-07-24:
Yes, this is still a problem. 
It can be replicated with Windows 7 basic theme, or you can enable the "disable disable scaling on high DPI settings" compatibility setting.

expltu, 2022-01-25:
Change your screen scaling to 100%
[image]