AgOpenGPS-Official/AgOpenGPS #52: I cannot build the solution.

TheRobertRalph, 2018-12-12:
I downloaded the project. I am excited to use it.

When I open the solution in Visual Studio, I then try to build and get errors.

They are all similar so I must be missing something.

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1078	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1083	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1091	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1096	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1104	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1109	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1117	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1122	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1130	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1135	Active
Error	CS1002	; expected	AgOpen	C:\Users\bob\Desktop\AgOpen-master\AgOpen-master\AgOpen_Dev\SourceCode\GPS\Forms\FormGPS.cs	1143	Active

```

farmerbriantee, 2018-12-17:
Just downloaded it and compiled and ran fine. Perhaps try downloading again, and try a Rebuild All.

TheRobertRalph, 2018-12-17:
Ok, got it to work with an install of VS and a new download. Thank you.

closed 2018-12-17 by TheRobertRalph