AgOpenGPS-Official/AgOpenGPS #59: Sections turning on and off randomly

charlesallis, 2019-03-24:
As I drive, sometimes the sections will flicker on and off for a short period of time, this especially happens when one section is off, the other one will flicker on and off with it when it should be on.

apmartin00, 2019-03-25:
This was an issue I had, did think there was an AOG Forum post about it.
Basically, check your GPS sentences coming in, should be just 1 line, if more need to check gps NMEA output setup and adjust accordingly, RMA 5hz is about minimum setting that works. Otherwise post a query on Combine forum.

mtz8302, 2019-04-11:
AOG needs only NMEA $GxGGA and $GxVTG.
It parses 1 sentence per loop, if there are too many sentences that are not usefull it doesn`t get a new position. So speed is 0, sections turn off.
Solution: configure your GPS to send only GGA and VTG