AgOpenGPS-Official/AgOpenGPS #60: Maximum Internal Boundaries in a Field - too low?

martini097, 2019-04-25:
Is there a rationale behind the limit of 6 boundaries?

I went to add to add some boundaries into a field to show where some irrigation hydrants [9 hydrants] are in a paddock, but ran into the 6 boundary limit. Changing this line let me add more, 
https://github.com/farmerbriantee/AgOpenGPS/blob/f8b3b0f1a385520e6769703fc6ae79d3dbd77c4f/AgOpenGPS/SourceCode/GPS/Forms/FormGPS.cs#L31 
however, the `FormBoundary.cs` seems to be limited to displaying just 6 boundaries still.

Would it be possible to have unlimited boundaries, with the form auto-filling/updating?
(side question: what is the minimum boundary size allowed?)

farmerbriantee, 2019-07-18:
Not sure about unlimited, but how many realistically would be sufficient. They have to be created before the program starts - a limitation in the programming unfortunately

martini097, 2019-08-08:
10 should probably do it, if that's feasible.

Jmaydavid, 2019-08-09:
I agree 10 would be really good. Great for spray-lines. Though tricking it with tool offset works well enough too.

farmerbriantee, 2019-09-03:
Daniel P has been working on dynamic boundaries, hopefully once we get this resource thing sorted, a lot more pulls can be done successfully.

closed 2019-09-03 by farmerbriantee