AgOpenGPS-Official/AgOpenGPS #64: [FR] Add feature: fertilizer spreader

hagre, 2019-07-29:
Would it be possible to add a different/larger marking to show spreaded fertilizer? Timing/Distance to the vehicle of the has to be adjusted as well.
I am thinking of such an equipment https://rauch.de/english/fertiliser-spreaders/mds/mds-17.1-19.1/index.html

ArBe-AU, 2019-08-05:
Hagre,

Can you provide some more information - alternatively ask your question in the AgOpenGPS thread of the Combine Forum https://www.thecombineforum.com/forums/31-technology/278810-agopengps.html

For the equipment in the link you provided I would make the section width the width the spreader is able to spread the product you are using - according the the website 10 - 18m.

If that doesn't help either post your question in the Combine Forum or provide more details here.

hagre, 2019-08-05:
Thank you for your replay.

I will post this and further requests on the forum.

Yes, the section size should be up to 15m in my case, but i can use (as a walkaround) more then one section to simulate the full width. The real problem is to mark the backwards sprading area (segment of a ellipse/circle - a triangle could be a good approximation) and calculate the right time to switch the sections on/off. 

Advanced thought ->show the reduced/different intensity of the fertilizer on the edge oft the spreading area of each disc in a differnet color/shading. The spreading areas are normaly overlaping in the middle of the two spreader discs and between the different tracks. To get a better sprading pattern we are driving on tacks with half the max sprading distance (see picture on link - "Abbildung 4" and "Abbildung 5").  An additionaly feature woul be the display of border and boundary spreading (limits the sprading width for one side (left) - smaler triangle.

A good discription and pictures are in german at https://www.dlg.org/de/landwirtschaft/themen/technik/technik-in-der-pflanzenproduktion/dlg-merkblatt-410/

hagre, 2019-08-05:
Link: https://www.thecombineforum.com/forums/31-technology/278810-agopengps-392.html

farmerbriantee, 2019-09-03:
Have suspended all pulls for now till we get the procedure and resource issues fixed

closed 2019-09-03 by farmerbriantee