AgOpenGPS-Official/AgOpenGPS #65: ABCurve is not parallel curve

quentinliao, 2019-08-09:


Jmaydavid, 2019-08-09:
This would be a really amazing feature... Can't think of much differentiating from the big guys after this...
Maybe as a sub-function within AB curve? So the default AB curve could assume non-circular geometry, which is realistic of most peoples paddocks! But have a button that switches the assumptions of the code to perfect circular geometry so that all AB curves become parallel? Would work perfectly for centre pivots. And use the snap to line to miss the wheel tracks. We would personally pay for someone to code this function...

Jmaydavid, 2019-08-09:
No idea how complex this would be... but the way JD used to do it in their parallel curves (think they called it pivot pro? A ridiculously expensive add-on) was to set the implement off the wheel track/start of curve (say 100mm) at one end of the pivot track, press point A, then drive round to the other end of the pivot track, set the implement at the exact same distance (100mm), and hit point B. Then I guess with that, computer finds the radii of the arc that intersects both points, and you've got perfect parallel curves?

Jmaydavid, 2019-08-27:
Well a single AB curve is all that's needed for pivots because the whole paddock works off that same curve/circle. For S curves, probably not. But I mean, the current AB curve works quite well for S-like curves from my testing.

quentinliao, 2019-09-03:
I tried a lot of ways to achieve parallel curves, but they all failed. Without turning radius control, the vehicle would not be able to drive on the generated curves. But I still haven't found a way to add turning radius control.

Jmaydavid, 2019-09-03:
Yeah. I've started realising how big this little side venture is... Apparently a Daniel fellow has found/is finding a solution. Would be very interested to see how he does it. 
[image]

farmerbriantee, 2019-09-03:
Coming to AgOpenGPS shortly!

closed 2019-09-03 by farmerbriantee