AgOpenGPS-Official/AgOpenGPS #82: SIM 868 GPS/GSM module serial connection

vanvince, 2019-10-04:
Hi, 

I'm trying to connect a GPS/GSM module to AgOpenGPS through one of my laptop USB port. The module is SIM868 GSM/GPS/GNSS hat by Waveshare. 
If I use this device with a simple puTTY serial connection it works perfectly fine and eventually gets GPS FIX on the other hand when I connect it to AgOpenGPS by clicking on "PORTS" and put the correct COM port, baud-rate and NMEA rate (1Hz) there is NO CONNECTION. I can read only the "Initial settings" message inside the text box.
Does anyone know how to solve this issue? Is it possible to connect this kind of module to AgOpenGPS? 

Thank you in advance.

richardklasens, 2019-10-04:
Does the receiver output nmea data?

Met vriendelijke groet,
 
Richard Klasens

vanvince, 2019-10-04:
@richardklasens yeah of course. 
Normally I send to this module an AT command and receive a response over the serial port. To obtain NMEA data I send this command sequence: 
AT+CGNSPWR=1 (to power on the GNSS module)
AT+CGNSTST=1 (to start receiving NMEA data) 
from now on it continues to send NMEA data through the serial port.

farmerbriantee, 2019-10-04:
Is it sending a GPGGA sentence? Can you take a screen shot of the Serial Ports form, what is in the box that shows the gps sentence?

farmerbriantee, 2019-10-04:
BTW 1 Hz is not much use other then to just show where you are, minumum 5 Hz required

vanvince, 2019-10-07:
@farmerbriantee  1Hz is module basic setting. 
Here there is an example of messages from and to module (I'm in my office so there is no GPS fix).
[image]
As we can see in the picture above the module gives a GNgga sentence and not GPgga. Maybe this is my issue. 
What can I do? Is there any way to change GNGGA to GPGGA? 

By the way, I have an ardusimple RTK module too. Can I use it? Are there any particular settings to do so?

best regards, 
Vincenzo Cristiano

aortner, 2019-10-08:
you should no use that module if you have a rtk from ardusimple.
just use ucenter to setup the module.

vanvince, 2019-10-08:
@aortner yeah I know but I wanted to test this module too. Anyway, ardusimple RTK works fine and it was extremely easy to use with AgOpenGPS

farmerbriantee, 2019-10-09:
AgOpenGPS will decode both GNGGA and GPGGA so that should not be a problem.

farmerbriantee, 2019-10-17:
please post your question in the forum... [https://agopengps.discourse.group/c/gps-rtk-gnss](url)

closed 2019-10-17 by farmerbriantee