AgOpenGPS-Official/AgOpenGPS #99: Use Nuget for dependent packages.

buttontate, 2019-10-17:
Looks like there are some dependences hard referenced in the solution. I was able to use Nuget for almost all of them. This would enable upgrading package references easily when they are released.

farmerbriantee, 2019-10-17:
This project is for people learning and just starting out. Trying to keep things as simple as possible, hope this makes sense

farmerbriantee, 2019-10-17:
I guess its should vs could.

buttontate, 2019-10-17:
I see your point and I'm just suggesting things to make things easier. 

Nuget package source is included by default in visual studio so it's no additional setup. The majority of .net examples on the web reference packages so they don't have to include the references in the repo or if its an article they don't have to host the actual references to download.

Just a thought.

closed 2019-10-23 by buttontate