AgOpenGPS-Official/AgOpenGPS #114: Google Maps Aerial Imaging

zpertee, 2019-11-14:
Hello!  Really enjoying your project!!  I'm actively looking at using aerial imaging with the application to replace the floor jpeg.  I have a couple of ways of doing this including drone imaging and using Google Maps API (https://medium.com/techtrument/how-many-miles-are-in-a-pixel-a0baf4611fff).  Are there any other requirements or can I simply replace floor.jpg with my new image?  Seems like there may need to be something to determine positioning on top of the image?  I'm interested in understanding the requirements and then I can run with the programming from there.  Thanks.

buttontate, 2019-11-15:
Hey @zpertee have you created a feature request in [discourse](https://agopengps.discourse.group/c/feature-requests)? This is where you are going to get your requirements and ideas.

zpertee, 2019-11-15:
Ok, thanks.  I will post there.  I was uncertain if the user space and the dev space were one and the same or not.

buttontate, 2019-11-15:
They discourse group is fairly new. Since there's not many developers there's not really a dev space. There's actually a call tomorrow to discuss topics. It would be nice to gather requirements in discourse and somehow create a development card for it. Cards can be feature branches and tracked/worked on that way. 

But, since you are here. Would you plan on being able to enable/disable the map? I imagine you have to plan on there being no internet connectivity.

zpertee, 2019-11-15:
@buttontate So, to be honest I've thought about this and come up with a handful of possibilities, but have not yet fully vetted my thoughts.  What I would like to try first is Google provides a REST API for retrieving static aerial images.  The way that it works is you give it a coordinate for the center point of the map and a zoom level and it returns an image of a given size.  When I get some down time I need to carefully review the mathematics and API, but roughly my thought is that for a rectangular parcel of land I can input the coordinates of the 4 corners and it will automatically calculate the center coordinate and the maximum zoom level that contains my parcel.  Assuming I'm able to gather enough of the right info it wouldn't be too much harder to crop that image down further to just my parcel.  

I see that the project has a floor.png file that is distributed.  If I basically overwrote that file with my Google Maps photo I'd potentially be part of the way there, but I can only assume not completely.  I've started a standalone app to work with retrieving the Google Maps image.  I'll likely take that to the point of reading the GPS data and moving a circle around to show my current position.  Other than the initial image download this will all be offline.  

I'm still new to AgOpenGPS and have not yet used it for real work yet, but it appears like you can provide a kml of the property boundary.  My suspected work flow is you import the kml and at that point the system detects the needed GPS coordinates and zoom level, reaches out to Google to download the image, crops it (etc.), and sticks the image in a special folder.  When you load that field if the custom photo exists it uses it and if not it sticks with floor.png.  I'll have to figure out how you guys handle positioning on top of the floor.png today and do something similar.

buttontate, 2019-11-15:
@zpertee  I like where you are going with this. Let me throw this at you. 

What about if the management of the floor was outside of AgOpenGps? Right now there's no automated way to transfer field/implement/tractor configuration besides putting them on removable storage and moving them to another device. 

I'd like to be able to create a web server that's dockerized so can run virtually anywhere (home server, azure, aws, gcp). You could have cloud have a storage provider that would store configuration information so you could just sync it to other devices. With a web interface you could enrich the fields (add the floor) or other metadata.

Honestly if you had the web server you could share positioning data between tractors. I would have to test latency but its doable.

zpertee, 2019-11-15:
@buttontate That definitely seems like the way to roll.  The last couple years I've been working significantly with docker, azure, and cloud messaging for an unrelated client project and I've become a big fan.  I really like your vision of a centralized endpoint and the sky is the limit of what you could build on top of this type of platform moving forward.

buttontate, 2019-11-15:
Well... nothin' to it but to do it. 

I'd like to get @farmerbriantee blessing on the matter as it would take some integration work. I do believe that some supporting infrastructure would be a really good thing. 

Lets create a repo and start creating some issues to start the talks of tech stack and objectives?

zpertee, 2019-11-15:
Sounds good, I concur.

buttontate, 2019-11-15:
@zpertee https://github.com/buttontate/AgOpenGPS-Server

farmerbriantee, 2019-11-18:
If you could figure out how to use GE as a background, that would be AWESOME!!!!! The image used is just a texture repeated to look like a surface, quite simple but gives some detail to see the surface better as you drive. So many applications use GE as a background, it is a nice feature. About 5 or 6 years ago you could do it with winforms, but google changed the API and the functions no longer worked. I haven't been able to find a working example that works with winforms and the new API. 

Another possibility is to extract a GE image by coordinates on the corners. The challenge however is there is limited texture size on many video cards with openGL.

zpertee, 2019-11-18:
Thanks for the feedback!  I have a chunk of availability from grads school between second week in December to second week in January where I hope to be able to dedicate some time to this.  We'll see how far I can get.  At least at the surface, it does not appear to be too difficult.

farmerbriantee, 2019-11-18:
Oh the surface is a bit bumpy :) But if there is anything i can help with please don;t hesitate to ask. If there is one thing i could suggest, please put this topic as a new request on the discourse forum. then it won't get lost.

closed 2021-04-19 by farmerbriantee