AgOpenGPS-Official/AgOpenGPS #123: No namespace Properties building AgOpenGPS in Visual Studio Code

vanvince, 2019-11-27:
HI,

I'm trying to build AgOpenGPS through Visual Studio Code (VSC). I installed everything needed for this project but when I run:

`C:\Windows\Microsoft.NET\Framework\v3.5\csc.exe /t:exe /out:AgOpenGPS.exe .\Program.cs `

into the terminal it shows me back this:

```
Microsoft (R) Visual C# 2008 Compiler version 3.5.30729.9034
for Microsoft (R) .NET Framework version 3.5
Copyright (C) Microsoft Corporation. All rights reserved.

Program.cs(1,17): error CS0234: The type or namespace name 'Properties' does not exist in the namespace 'AgOpenGPS' (are you missing an assembly reference?)
```

On the web, I found out that I should add the references through 

`-r:"\Path\of\references"`

option to the command shown above. But the missing namespace is one of AgOpenGPS itself, right?
Honestly, I don't know what to do, any suggestions?

buttontate, 2019-11-27:
Hey @vanvince. VS Code might be a problem for this particular project. It uses resource files for its language translations which Visual Studio is the only program that has an editor that I know of.

Download Visual Studio, its free and you'll run into a lot less problems with the debugger.
https://visualstudio.microsoft.com/vs/

vanvince, 2019-11-28:
Hi @buttontate ,

yeah, you're right! I already thought about it yesterday but I wanted to try with VSC. What a failure LOL 

By the way, building AgOpenGPS I found another issue. Everything goes right until I launch the exe.  After that nothing happened.  So, I found the older version on my PC and comparing Programm.cs file I found out some comments from line 24 to line 44 concerning 

`RegistryKey regKey = Registry.CurrentUser.OpenSubKey(@"SOFTWARE\AgOpenGPS");`

I commented all of these lines and build again. Now it works fine.

farmerbriantee, 2019-12-08:
No idea why that would happen, but if you compile the latest version, do you stll have issues? The best place to post these problems is on the AOG forum. https://agopengps.discourse.group/

closed 2020-01-10 by farmerbriantee