AgOpenGPS-Official/AgOpenGPS #170: Is it wrong calculating angular velocity in CABLine.cs ?

luckydaxiang, 2020-06-17:
The formula  for angular velocity is ω = v / r .
In  CABLine.cs line 587, I think it should be :  `angVel = 0.277777 * mf.pn.speed * (Math.Tan(glm.toRadians(steerAngleAB))) / mf.vehicle.wheelbase;`
[image]
Can you explain to me ? Thank you .

yuyongji1987, 2020-12-09:
I have the same question. Did you get the answer?

closed 2021-04-19 by farmerbriantee

mirh, 2021-04-19:
https://physics.stackexchange.com/questions/112301/calculating-the-rate-at-which-a-car-turns
Mhh.. Indeed something smells off. Perhaps there was some mix up with the uniform circular motion formula?

farmerbriantee, 2021-04-20:
> The formula for angular velocity is ω = v / r .
> In CABLine.cs line 587, I think it should be : `angVel = 0.277777 * mf.pn.speed * (Math.Tan(glm.toRadians(steerAngleAB))) / mf.vehicle.wheelbase;`
> [image]
> Can you explain to me ? Thank you .

Sure can, angvel isn't actually used and adding pi several years ago made it easy for setting the value against a slider - none of which is used anymore anyway.