AgOpenGPS-Official/AgOpenGPS #171: Add the arduino code back to the repository

charlesvdv, 2020-06-20:
Hello, 

I setup AgOpenGPS for my father a almost two years ago. Since then, a lot has changed and I would like to upgrade to the newest version. During those two years, I noticed that the pin mapping on the arduino changed. So a few months ago, I setup a fork of the repository to track the different pin mapping ([my fork](https://github.com/charlesvdv/AgOpenGPS/tree/production)) in git to make it easier to update. I just had to pull from your repo and git would tell me if there is a conflict. 

Now, I just noticed that the arduino code is no longer in the repo which makes it impossible for me to track those changes (or at least makes it really really difficult). Would it be possible to add the arduino code back? This issue also relates to #165 

Thanks a lot!

mirh, 2020-06-20:
I mean, it kinda has its sense to keep it separated from the main application. They really are different things. 
On the other hand, just having it in the release section.. Kills any collaboration attempt. 

I guess like it should become its own standalone repository.

charlesvdv, 2020-06-21:
Personally, it would make a lot of sense to keep it in the same repository as the two are thighly coupled. If a change happens in the firmware (aka. arduino code) and there is a high likelyhood that the the application code has to change.  

But in the end, if the arduino code is added back in the repository or in another repository, it would already make my life much more easier..

closed 2021-04-19 by farmerbriantee

mirh, 2021-04-19:
This wasn't solved at all.

mirh, 2021-04-20:
I'm glad of it.. 
But maybe a bit more participation wouldn't hurt. That's the spirit of the community. 

I get github isn't some instant messaging app or the forums, but exactly for this reason you can have "expert" discussions about code (and code practices) here.

richardklasens, 2021-04-20:
Arduino code is now in master