AgOpenGPS-Official/AgOpenGPS #172: Should the hitch arm length belong to tractor?

svenn71, 2020-07-06:
Currently, when setting up an implement attached to the 3-point hitch, the length of the hitch arms is set in the implement setup. I then have to have different 5-section sprayers depending on which tractor it is mounted on. Maybe there are technical reasons for this that I do not know about.
I also find the same issue with my towed combined fertilizer/seeder. The distance from the applied area to the eye of the tow point on the implement is the same, but the distance from the tow hitch to the pivot line is different on all my tractors. On the setup screen for the towed implement the distance is clearly divided in two, one on the tractor side for the tow hitch and one on the implement side for the tow length. Moving tractor measurements to the tractor config would be a real customer added value.

apmartin00, 2020-07-06:
This has been dealt with previously. Each setup is a combo of tractor/implement so record it as such. If your sprayer goes on 5 diferent tractors you would have 5 tractor/ sprayer setups with unique names.
Best to ask questions like this on the agopengps discourse site as more people will see it there.

richardklasens, 2020-07-08:
Hitch length needs to be a part of the tractor. 
Implement length needs to be inserted in implement settings. Then we have it independent for tractor/implement.

closed 2021-04-19 by farmerbriantee