AgOpenGPS-Official/AgOpenGPS #173: Label buttons

mirh, 2020-07-26:
It's really so hard to try to guess what each button does just from the icon.

closed 2021-04-19 by farmerbriantee

mirh, 2021-04-19:
There's still **NO** label whatsoever.

richardklasens, 2021-04-19:
Labels needs translation in 35 languages... when docs/videos are ready.. it will be inserted. We don’t start labeling buttons in program.

mirh, 2021-04-19:
An english label is already better than no labels?

I reckon V5 icons are somewhat easier to understand, but if I didn't knew most buttons already, the only way I'd have to be sure of their functionality would be reading the source code.

richardklasens, 2021-04-19:
Programmers choice..

mirh, 2021-04-20:
It doesn't make sense *logically*, and for the user?
Even chinese would be better than having to scrape the code. 

Btw, when I say label, I mean some tooltip when the cursor has been hoovering on a button for some number of seconds. 
Not horrendously mixing the sleek UI with text.

richardklasens, 2021-04-20:
I understand what you’re saying but now try tooltips and hovering on a touchscreen :)

mirh, 2021-04-20:
Microsoft UX guidelines are just that hover is nice and dandy (you can even see this with the tooltip when you move the pointer on top of the date on the bottom right of the screen), just don't make it "necessary" for core functionality. 
This seems the case here. 

Besides, press and hold should be a relatively common touch equivalent (even though I guess it depends on the specific drivers and whatnot).

mirh, 2023-04-21:
I suppose this could have been considered fixed with #227. 
It's even smarter than what I had in mind, since very obviously it wouldn't have been the most of intuitiveness with touch controls Also, given the absolute deluge of buttons on most given windows, it avoids any unfortunate happenstance where the random position of the cursor could have caused an unwanted popup. 

***Still***, I feel like at least for the help icon itself, that could still be implemented. 
Like.. I can *see* more naive people \*fearing\* to click random buttons (even if it's a literal question mark).