AgOpenGPS-Official/AgOpenGPS #177: Can't switch back to Simulator mode after setup

mirh, 2020-09-02:
I think the steps to reproduce are:
* configure your GPS receiver and connect it to AOG
* close AOG
* start it again and wait for precision to get good enough the "I'm lost" marker is no more
* get back to the "Serial Ports" window and disconnect it
* switch on simulator and click to restart

It's like the last two steps never happened.

richardklasens, 2020-09-02:
You have to disconnect the USB cable, otherwise it will reconnect to the latest chosen serialport

mirh, 2020-09-02:
I get that the connect/disconnect buttons may only be temporary to the local session.. But if I enable the simulator and I restart the program to get that, I am expecting that.

closed 2021-04-19 by farmerbriantee

richardklasens, 2021-04-19:
This is solved by using AgIO. Close it, turn on sim. As soon as agIO is running and is sending position data, sim is turned off again