AgOpenGPS-Official/AgOpenGPS #178: Problems with maps "very far away"

mirh, 2020-09-24:
Say I open a .kml "real life" field while I'm in the simulator, or I open a test field I made in the simulator while I'm using the GPS... 
The operation takes an atrociously long time, and the field either doesn't load at all or the ground texture becomes black.

apmartin00, 2020-09-24:
This is not a code issue, more an user issue. delete your test fields, set the sim co-ordinates close to your home. then try again.
better to search or ask questions on https://agopengps.discourse.group/ rather than here.

mirh, 2020-09-25:
Ehrm, it's not a question though (just like all my other bugs). 
The program should either work gracefully with whatever you throw at it (you get moved to the new coordinates for example, and it doesn't glitch out), or if that's not doable it should return an error. 
It's really far from user-friendly otherwise.

closed 2021-04-19 by farmerbriantee

richardklasens, 2021-04-19:
Fixed in v5

mirh, 2023-03-19:
This is still a problem actually (and assuming the starting coordinates for the simulation mode are brian's house IIRC, ironically enough it would be virtually impossible for him to spot the problem). 

Spraying weedkiller didn't really require anything fancy today, so I just opened the first random field that was selected (to enable the action buttons) and then I used manual section control to account for whatever land had already been treated. 
Somehow the moment I started to move, hell broke loose and I could see the on-screen tractor moving back and forth like a madman. It was legit *jumping* every frame or so, and the line it was supposed to trace seemed absolutely drunk. 

Long story short, after banging my head for a hour, it turned the culprit was my "dsfsjdfk" test field that in the hurry of the moment I had continued/carried from the simulation mode (that I had been using right until minutes before). 
Once I created a brand new one, everything turned back to perfect normality.

richardklasens, 2023-03-21:
Maybe add an error message if you're further away then 50km or so.. 
(Or just don't open test fields... in the drop-down menu of the fields you can actually see you're far far away from it...

*if you open fields when SIM is active it transfers the field offset data to the sim coordinates.

mirh, 2023-03-22:
I suppose both are good (and simple) ideas, but I cannot avoid to wonder if there isn't an even bigger silver lining here. 
Like, sure thing, keeping in memory a map thousands and thousands of kilometres wide isn't a walk in the park... but why, really?

Why is it that only the "applied area" overlay/layer is affected, and why should the "empty in-between" matter to anything if my zoom level is currently set to show me a section of the world about 50x50m big (and it's not like my tractor can split in two)?

mirh, 2023-04-18:
Until this is fixed (my old man *really* has a short memory) the issue should be reopened isn't it. 

Another idea from today's misadventure, is that the current field should get automatically  closed on exit from SIM mode.

mirh, 2023-05-22:
Uh, now that I think to it..
Remind me again why the simulator mode even needs the concept of *fields* to begin with?

Like, as far as I can tell.. you just use it to test the equipment without having to actually move yourself. 
It's not like "teleporting behind your backyard and doing some virtual stuff on the ground" is unthinkable, but I just cannot see what the purpose could even be.

mirh, 2024-04-19:
This was fixed for good in https://github.com/farmerbriantee/AgOpenGPS/pull/351/commits/e21189360e24018d9e722fc5f22f73063be078a6