AgOpenGPS-Official/AgOpenGPS #181: Same KML is loaded with different sizes

mirh, 2020-11-11:
As also reported [here](https://agopengps.discourse.group/t/wrong-file-area-size-of-google-earth-kml-imported-boundary/1726).. But I have a [testcase](https://github.com/farmerbriantee/AgOpenGPS/files/5526806/Fields.zip) on my side. 

Basically, if you open the KML on a fresh AOG instance with the first field loaded, it's the right size of about 1 hectare and half. 
If you open the same thing while you have the second field in memory, it's something like *77*.

closed 2021-04-19 by farmerbriantee

mirh, 2021-04-19:
I'm not even able to open the old files anymore. 
The .kml throws a "Boundary Lines File Corrupt - But Field Is Loaded" error. 
And one of the two fields doesn't show anything.

richardklasens, 2021-04-19:
You should be able to open fields from 4.3

mirh, 2021-04-20:
One opens, the other doesn't. 
They are attached in the second link above, if it wasn't clear.

richardklasens, 2021-04-20:
Open kml or open field txt from 4.3?