AgOpenGPS-Official/AgOpenGPS #186: Offline multiplier check is backwards

Kython89, 2021-01-23:
`if (distanceFromCurrentLine < 1.0)
                goalPointDistance += distanceFromCurrentLine * goalPointDistance * mf.vehicle.goalPointDistanceMultiplier;
            else
                goalPointDistance += goalPointDistance * mf.vehicle.goalPointDistanceMultiplier;
`
The distance check seems backwards.  If the intent is for the offline multiplier to result in a larger lookahead when the distance from the line is larger (> 1 m) then this is backwards.

The actual behavior is:
Distance == 0 then it has no effect.  
As the distance builds the look ahead is increased.
As the distance builds past 1 then the offline multiplier has no effect.

P00qwerty, 2021-01-23:
its supposed to work this way, as you want to steer faster to the line the further away you are.

Kython89, 2021-01-23:
Sorry.  You're right I was thinking about this backwards.

closed 2021-01-23 by Kython89

Kython89, 2021-01-23:
Now I'm confused again.  In this video https://www.youtube.com/watch?v=tFYXUu6782Y Brian says that the offline look ahead multiplier makes the look ahead distance longer based on how far you are from the path.  I guess my thought and experience is that trying to get on the line with a small look ahead distance results in oscillation where a larger distance allows you to smoothly acquire the line.  Or does this only apply when you are within 1 meter from the line?