AgOpenGPS-Official/AgOpenGPS #192: distancefromline removed from AutoSteer PGN

MattWoodhead, 2021-04-02:
Those of us who use external lightbars cannot upgrade to V5 as the 'distance from line' value has been removed from the AutoSteer PGN and is not present in any of the messages that have changed from V4 to V5.

Is it possible to add it into a new PGN, or maybe in the two spare bytes in the Machine Data message?

mtz8302, 2021-04-21:
Instead of distance from line, you might process the send steer angle setpoint. This will only work with straight lines, as you don't have actual steer angle of your tractor to compare

MattWoodhead, 2021-04-21:
> Instead of distance from line, you might process the send steer angle setpoint. This will only work with straight lines, as you don't have actual steer angle of your tractor to compare

I have attempted this here: [https://github.com/MattWoodhead/AutosteerPCBv2LB/tree/AOG-5.0.0beta](https://github.com/MattWoodhead/AutosteerPCBv2LB/tree/AOG-5.0.0beta). I logged the requested steer angle and distancefromlime value with the 4.3.10 simulator in different scenarios to generate a relationship between the steer angle and the cross track error. It just about works (mathematically) if you are going in a straight line and need to correct, but it becomes inaccurate if you a following a curve usg contour or similar as the demanded steer angle from AOG is different for the same XTE as AOG has the advantage of knowing what is coming. I haven't had time to sit on a tractor or in front of the simulator on V5 to see if it is acceptable for real world use.

Hopefully we can aadaddadd tththethe XTE back in again: https://agopengps.discourse.group/t/external-light-bar/830/51

farmerbriantee, 2021-04-24:
autosteer PGN delivers distance at 2cm per count centered at 127. 
255 is invalid or off. 
byte 10 of pgn 254

closed 2021-04-24 by farmerbriantee