AgOpenGPS-Official/AgOpenGPS #194: Crash with custom dpi

mirh, 2021-04-09:
As I had already complained in https://github.com/farmerbriantee/AgOpenGPS/issues/48#issuecomment-663770707
```
Application: AgOpenGPS.exe
Framework Version: v4.0.30319
Description: The process was terminated due to an unhandled exception.
Exception Info: System.NullReferenceException
   at AgOpenGPS.FormGPS.LineUpManualBtns()
   at AgOpenGPS.FormGPS.FixPanelsAndMenus(Boolean)
   at AgOpenGPS.FormGPS.FormGPS_Resize(System.Object, System.EventArgs)
   at System.Windows.Forms.Control.OnResize(System.EventArgs)
   at System.Windows.Forms.Form.OnResize(System.EventArgs)
   at System.Windows.Forms.Control.OnSizeChanged(System.EventArgs)
   at System.Windows.Forms.Control.UpdateBounds(Int32, Int32, Int32, Int32, Int32, Int32)
   at System.Windows.Forms.Control.UpdateBounds(Int32, Int32, Int32, Int32)
   at System.Windows.Forms.Control.SetBoundsCore(Int32, Int32, Int32, Int32, System.Windows.Forms.BoundsSpecified)
   at System.Windows.Forms.Form.SetBoundsCore(Int32, Int32, Int32, Int32, System.Windows.Forms.BoundsSpecified)
   at System.Windows.Forms.Control.ScaleControl(System.Drawing.SizeF, System.Windows.Forms.BoundsSpecified)
   at System.Windows.Forms.ScrollableControl.ScaleControl(System.Drawing.SizeF, System.Windows.Forms.BoundsSpecified)
   at System.Windows.Forms.Form.ScaleControl(System.Drawing.SizeF, System.Windows.Forms.BoundsSpecified)
   at System.Windows.Forms.Control.ScaleControl(System.Drawing.SizeF, System.Drawing.SizeF, System.Windows.Forms.Control)
   at System.Windows.Forms.ContainerControl.Scale(System.Drawing.SizeF, System.Drawing.SizeF, System.Windows.Forms.Control)
   at System.Windows.Forms.ContainerControl.PerformAutoScale(Boolean, Boolean)
   at System.Windows.Forms.ContainerControl.PerformNeededAutoScaleOnLayout()
   at System.Windows.Forms.ContainerControl.OnLayoutResuming(Boolean)
   at System.Windows.Forms.Control.ResumeLayout(Boolean)
   at AgOpenGPS.FormGPS.InitializeComponent()
   at AgOpenGPS.FormGPS..ctor()
   at AgOpenGPS.Program.Main()
```
I don't know if it's my having updated to version 5.0.0.3, or to W10, this time only custom dpi factors (as opposed to even the standard fixed "25%" increments) seems problematic

mirh, 2021-04-19:
Hello?? This is a serious bug that is gonna affect a lot of people.

richardklasens, 2021-04-19:
How about this in the latest source?

mirh, 2021-04-19:
Yes this bug is still in the latest sources. 
I don't know what's the point of closing them all regardless of their merit.

farmerbriantee, 2021-04-19:
What is the solution? Any idea?

mirh, 2021-04-20:
Nope, but considering it has workarounds, no hurry. 
I suppose you have far worse nuts to crack on release week. 

Just.. read stuff before mass nuking it from orbit.

farmerbriantee, 2021-04-20:
I did, others did, no problem

farmerbriantee, 2021-04-20:
issues were removed because we are on v5 and no one will be fixing v4, 

If they are actual issues still, they will magically appear again and can be dealt with

farmerbriantee, 2021-04-20:
The other really helpful thing is coming up with a solution to a problem no one on the development team can replicate. If we can't make it crash, we can't fix a problem that doesn't exist.

mirh, 2021-04-20:
Normal practice is that you bump the thread and ask the author for an update, and if you get no answer after a given number of days you close it. 
Then, I get that v5 is a big beefy new version, but it's not like a total rewrite. And even totally legit and "abstract" questions like #170 have been sent to oblivion. 

Anyway, this is so weird since I can reproduce it on all my pcs instead. 
If you are on W10, did you set the custom scaling to 115% for example?

farmerbriantee, 2021-04-20:
So i tried custom scaling of 157, 167, and normal scaling of 150, 175, 200. Works just fine.

farmerbriantee, 2021-04-20:
I should mention there is a discussion forum which gets read a lot. This issues page, not very much.

mirh, 2021-04-20:
I know... 
But my line of thought was always "that's for chit-chat", "this is for serious business". 

Anyhow, I really don't know what else I can add. 
W7 has latest .NET 4.8, while W10 has whatever comes out of the box with 20H2. 
The crash is just trying to open the program like at all, and together with #176 (which is supposedly no more) I no-joke spent an entire week banging my head on these crazy "pixels size" matters.

farmerbriantee, 2021-04-20:
> I know...
> But my line of thought was always "that's for chit-chat", "this is for serious business".
> 
If you want a question answered promptly and looked at, post it in the chit chat.  :)

Here is the answer to your angvel questions as well:

Sure can, angvel isn't actually used and adding 2pi several years ago made it easy for setting the value against a slider - none of which is used anymore anyway.

And also it isn't the actual steer angle, its the desired steer angle to get to the line - not what the vehicle is actually steering. Always remember AOG is a video game - it doesn't have to be correct or follow physics - it just needs to work. And even more confusing, it isn't even velocity, its the formula for acceleration.

expltu, 2022-01-25:
It is caused by custom scaling. Mine was set tio 105% - and the app was crashing. 
[image]

mirh, 2022-02-13:
QED.

mirh, 2023-04-12:
I'm once again pointing out this still happens.