AgOpenGPS-Official/AgOpenGPS #197: Basic auto steer code

cnoo8811, 2021-04-27:
I am trying to build a basic auto steer with only a Arduino nano, motor driver, wheel angle sensor and a steer switch. it is built now and the code from v 4.3.10 works with a little tweaking just wont read my wheel angle sensor. The sensor itself has 5v and a good ground to it and is sending back 1v when wheels are straight. The 1v is feed into port A0 on the Arduino. Just stuck and wondering if theirs something simple I need to change in the code to get this to read so I can go plant. Thanks

richardklasens, 2021-04-27:
Hi there. 

The code as is for v4.3 uses an ADS1115 as AD converter. It doesn’t use the arduino pins. For the future it’s better to post your questions on the discourse forum 

https://agopengps.discourse.group

Best Regards,

closed 2021-04-27 by richardklasens

cnoo8811, 2021-04-27:
I got a pmod ad2 digilent will that still work