AgOpenGPS-Official/AgOpenGPS #198: AOG Variable rate control

mappaturakiwiverona, 2021-05-05:
Dear Brian,
AOG v5 is working very well and looks like great!! Really impressive work done!!
I have a question, Is it plan to manage also variable rate prescriptions?
Is it possible to have a support from you to integrate https://github.com/Coffeetrac/AG_ASD_Host.git into AOG?
Maybe you have already solve it but I was not able to find a solution.

Thanks for helping me.
Best Regards
Davide M.

mirh, 2021-05-05:
[That](https://www.thecombineforum.com/threads/diy-autosteer-with-agopengps.296233/page-80#post-3307993)'s for Müller's ASD protocol. 
Which is not bad, but [this](https://github.com/SK21/AOG_RC) is more generic.

mappaturakiwiverona, 2021-05-05:
Thanks for replay,
I know the second one, but since I have to control a Muller one could you please help me to understand if it is possible to do it with the AOG v5 or I need to go back to the AOG version showed in the video?
To make it works, I need to set the Arduino COM in the AIO right?

In a YouTube video it is showed that with AOG is possible to do VR application using Headland JPG pictures, could be possible to do it also with AOG v5?

thanks for support
Davide M.

mirh, 2021-05-06:
According to a doc that I have found, Mueller uses either LH5000 or ASD, so you'd have to figure it out.
Anyhow, I don't really think that 2yo code is still gonna work with the last version... But there are people discussing that [right now](https://agopengps.discourse.group/t/asd-interface/4822). 

 AOG_RC does work with v5.

joaquinperaza, 2021-05-08:
@richardklasens I agree

closed 2021-05-08 by richardklasens