AgOpenGPS-Official/AgOpenGPS #217: discourse group cannot be found

svenn71, 2021-09-16:
when clicking on the link to the discussion group, discourse says that the group does not exist there.
Has something bad happened?

richardklasens, 2021-09-16:
We switched to a self hosted solution about a month ago. It was posted on the old forum and all telegram groups. 

Please use your credentials from the old discourse forum on our new link

https://discourse.agopengps.com

closed 2021-09-16 by richardklasens

mirh, 2021-09-16:
Why in the heaven is the forum set to private?

FarmerKevin, 2021-09-17:
Yes I could get on a few days ago, but not now.

mirh, 2021-09-30:
Fixed in #218 I guess, but there's still no explanation on why threads aren't public anymore.

richardklasens, 2021-10-01:
> Fixed in #218 I guess, but there's still no explanation on why threads aren't public anymore.

We had some problems with our old discourse plan, which forced us to turn this function on. It’s public again now. 

Sorry for the late reply