AgOpenGPS-Official/AgOpenGPS #306: Saving a field with the function 'Save As' gives an error when confirming

chrisriteco, 2023-03-25:
Saving a field with the function 'Save As' gives an error when confirming. Exception in FormSaveAs.cs on Line 125. (Is one space too many in the path), this is added on line 68, but goes wrong. The strange thing is that this problem is not in the 5.7.2 release and nothing has been changed in the file.

mirh, 2023-05-07:
The last git version has only one extra commit compared to 5.7.2 (unless you are talking about some of the alternative branches). 
And that's just a pretty dumb readme adjustment. 

Are you sure the problem couldn't be with whatever dev environment you have?

richardklasens, 2023-05-07:
We tried this in the release but I can’t recreate the issue here

chrisriteco, 2023-05-09:
Master branch checked out again and can't reproduce the problem.

closed 2023-05-09 by chrisriteco