AgOpenGPS-Official/AgOpenGPS #311: Nitpicks for steer wizard

mirh, 2023-04-24:
- [ ] there's no "IMU X or Y axis" option (in fact, is it normal that there's literally no control panel with all IMU settings together?)
  - [ ] btw did you ever think to some better wording for it, like "orientation" or "resting plane"? It has no obvious referent as of now
- [ ] "Invert Hydraulic Enable Relay" isn't exactly King's English, is it?
  - [ ] not that even "Invert Relays" in the config was saying much tbh (the help tooltip is 200% clear then, but it shouldn't be a must)
- [ ] it's nice that CPD calibration tries to automagically calculate its value.. but I got really mixed results from it (though even if I have the recommended RQH100030 sensor, I'm using the [A0 mod](https://discourse.agopengps.com/t/no-was-reading/7817/6) without a dedicated ADC so I'm unsure if this couldn't be on me)
  - [ ] and for all that is [explained](https://discourse.agopengps.com/t/how-counts-per-degree-work/4792) about its calculation, it would be nice if a line noted how AOG can tell the "real steering angle"
  - [ ] the wizard repeats the same process to calculate ackermann... is there any wisdom instead in the other [proposed](https://discourse.agopengps.com/t/autosteer-basic-configuration/8848) method?

gunicsba, 2023-04-27:
@mirh on 2 of the tractors I oversee we landed at 100 CPD using the TPS sensor. One guy almost got the 2.5V at the middle right as well.

Invert relays are basically relays being active high or active low.

These labels / information should be in the gStr file.

mirh, 2023-04-27:
Ok, so I guess that ~110 value I eventually settled with (and boy aren't there odd times when I can measure it ±5) is likely due to @v12cat's mod. 
Is there any possibility for it to be integrated/improved? Or are these kinds of problem exactly the reason you decided to settle with the ADS1115?
(yes I suppose these are a bit dumb questions, but even if it was to [improve](https://yihui.org/en/2019/05/rtfm-or-itfm/) the documentation they should be asked imo)

> Invert relays are basically relays being active high or active low.

Sure, there's no way you can have doubts once you read the explanation. 
Still, I wished the \*subject\* of the *inversion* was slightly more obvious (in particular, the button comes after two very self-explanatory icons, where the word means "direction" as opposed to "boolean state"). 

> These labels / information should be in the gStr file.

I mean.. labels by definition cannot go into the help information. They are the way you present the setting in the first place. Though yes, the majority of my pedantic points should go in there.

closed 2024-02-15 by farmerbriantee

mirh, 2024-04-18:
This wasn't fixed (and I understand it was kind of a mess, with mostly high-hanging grammatical navel gazing fruits)
But the situation is now even way worse given you [removed](https://github.com/farmerbriantee/AgOpenGPS/pull/384) the built-in help feature.