AgOpenGPS-Official/AgOpenGPS #312: Steer angle calculator uses normal direction assumptions

mirh, 2023-04-24:
I suppose there's really no way to avoid the "naiveness" AOG has when you restart from a standstill after having just gone reverse, and it needs some time to figure out that you are now actually going into the opposite direction (except perhaps tight integration with the IMU, but that's a story for another day).
But when you press the `🔴REC` button, with the only purpose of driving with a constant steer angle, it doesn't seem unreasonable for whatever the previous heading information was to be discarded.

Instead, I found myself over and over again at the end of my 15 meters turn, only to realize that the measuring got disengaged seconds after I started because it picked up the 180 flip in direction (it's not that hard to spot immediately when you let autosteer handle the constant turn, but if you are trying to keep the wheels steady manually then your attention is really only on the angle on the green level)

And this could be even worse in the steer wizard, because it's not that hard to interpret the "quick run" that you'd have with the tool as just the automatic algorithm being smart and already satisfied after a meter or two (and yes, the number you would get here is usually ludicrous enough that you should notice something is wrong.. but still).

mirh, 2023-04-29:
Actually, I found myself inconvenienced even while normally testing the autosteer steadiness (aka lots of back and forths). 
I totally get that there's just so much that you can do to foretell direction "in a vacuum" (IMU was disconnected).. but still, the software shouldn't treat the "final call" like the world *itself* had literally just spun around.

richardklasens, 2023-05-07:
This will all be solved in the next release hopefully. You can try the slowpoke branch to check if this gives you the right fix

closed 2023-05-07 by richardklasens