AgOpenGPS-Official/AgOpenGPS #313: IMU is always factored in even at its minimum setting

mirh, 2023-04-25:
Due to [reasons](https://github.com/farmerbriantee/AgOpenGPS_Boards/issues/22), I needed to "exclude" the IMU from being taken into account for guidance...
(and I just barely figured out that had to be the problem, because in my other machine without it I never had anything so blatantly off)

Yet, even with "IMU GPS Fusion" on its rightmost side, my sudden drifting/jerking problem when starting from a standstill persisted. 
...which I can still guess has its sense somehow, since starting from position zero would leave the GPS heading with only a pointwise frame of reference for the first half-second or something. So even the slightest breeze of an input from the IMU would still win here. 

Yet, you can see why this is a bit of an anti-feature. 
I mean, both because of my "particular situation", but also because logically it's kinda counter-intuitive. 
If the slider says "0 IMU", that should mean *0 IMU* (unless, I don't know, you want to leave the task of its complete cut off without ifs or buts to some separate button).
It shouldn't be that only if you neutralize its code in the Autosteer_USB ino, you can finally free yourself from its corrections.

richardklasens, 2023-05-07:
See comment in other imu issue

closed 2023-05-07 by richardklasens

mirh, 2023-05-07:
I suppose I may have infringed on the "one problem, one issue" rule (you start to get depressed after a while).. but why in the heaven did you close this?
IMU having no way to be removed from the equation still seems meaningful.

richardklasens, 2023-05-07:
> See comment in other imu issue

Because of this

mirh, 2023-05-07:
That's not an explanation at all, it's just a badly formatted copy-paste of my second comment here. 
If you consider that to be separate from my first post, it's certainly reasonable to "move elsewhere".. but it doesn't address in any shape or form the fact that the UI doesn't have any way to "ditch" the IMU information completely.

richardklasens, 2023-05-07:
You really have to do something about the way you write down your reactions. It seems you have absolutely no idea how to conversate in a normal way with a group of people who maintain this project just for learning, fun, helping farmers etc. 

If you need to rip off your imu due to its bad behavior, then you need to solve the bad behavior instead of almost forcing us to create a function to turn it off completely. 

For the last time.. in the coming release the problems people occur with IMU’s and driving directions, spinning tractors in the screen and such stuff should be solved. 

We do not going to add an IMU OFF button for your particular problem.. solve your problem.

now I need to find how to lock this conversation because I am done with the forced language you use in your comments