AgOpenGPS-Official/AgOpenGPS #318: Also tangentially related, that the IMU value (or if not any its graphical representation with the level/horizon at least) doesn't actually represent the "instantaneous signal" but just the last received value.

richardklasens, 2023-05-07:
Also tangentially related, that the IMU value (or if not any its graphical representation with the level/horizon at least) doesn't actually represent the "instantaneous signal" but just the last received value. 
If for whatever reason there are no more readings (because you disconnected the board, or because you just fried it entirely..) it stays stuck and doesn't reset back to 0.

_Originally posted by @mirh in https://github.com/farmerbriantee/AgOpenGPS/issues/313#issuecomment-1536260846_

mirh, 2023-05-07:
You use `>` for quotes, if you use more than 4 spaces it's interpreted as a code snippet. 
(also, the title should probably be something a bit shorter and coincise)

mirh, 2023-05-07:
~~I'd put more emphasis on the fact that the update loop is just fine lingering on stale data (or maybe, could it be described as the difference between using a "push" vs "pull" logic?) but you are the boss then~~
Ok I suppose it's kinda the chore to work with whatever the hardware, that could have whatever the independent refresh rates... so yeah maybe the point is just having to add a 1 second grace period or something like that.

closed 2024-02-15 by farmerbriantee