AgOpenGPS-Official/AgOpenGPS #319: Add more nuance to NTRIP server errors

mirh, 2023-05-10:
For the better part of last week, my trusty rtk corrections server was down. 
And I do know how to do a ping test, but not everyone is that shrewd. 
It would be nice if the `Connection Problem", "Not Connecting To Caster"` error could be a bit more specific. For instance:

* no answer at all on the specified server/port combo (maybe coupled with a check about internet connectivity working at all?)
* invalid credentials (error 401)
* resource/mountpoint/path not found (error 404)
* trimble also [seems](https://receiverhelp.trimble.com/alloy-gnss/en-us/ioConfig.html#IBSS/NTR) to define more status codes (602 and 604?) but idk really
* [something else](https://en.wikipedia.org/wiki/List_of_HTTP_status_codes) you think may be relevant?
* I did once solve an issue with the same aforementioned server, where my account wasn't authorized to have multiple clients connected at once (connection error 10035, in u-center) but I'm unsure how much specific this couldn't be

p.s. I also don't think it makes the most of sense to still have this error popup when you disconnect GPS and you are using the simulator

craisbeck, 2023-05-10:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

I did a bunch of troubleshooting with a pesky RTK connection this past winter and I thin the problem with your request is the errors and responses are different for different networks.  A connection failure on one system could be called Authentication Error and on the next system no error is reported and the next system a sign on failure.  My understanding is that there isn't a great deal of consistency across casters.

craisbeck, 2023-05-10:
Code it up.  It sounds promising.

I mean, I really don't think there's any way you can go wrong with HTTP error messages (I suppose there's the small dilemma of choosing whatever you want a SYN or ACK tcping, but it doesn't seem that fundamental)

Then.. uh, I get your concerns for unstandardization, but at worst that should only mean that you'll have to drop whatever knowledge you have in a particular case, if you cannot pin down the quirk to some unique technicality.
But it shouldn't be that hard imo (putting even aside that the likely custom codes I mentioned above, are so sparse and far between that the likelihood of collisions sounds minimal).

For instance, the server that I use identifies itself in the ICY 200 OK HTTP response as "GNSS Spider + version", like so<https://github.com/Road-Balance/ntrip_ros/blob/master/scripts/get-ntrip-source-table.py#L13> or this<https://github.com/ArduPilot/MAVProxy/pull/913>.

-

lansalot, 2023-05-10:
> I mean, I ***really*** don't think there's any way you can go wrong with HTTP error messages (I suppose there's the small dilemma of choosing whatever you want a SYN or ACK tcping, but it doesn't seem that fundamental)

I think your mistake is to assume all casters use standard http responses. And they just don't. I looked about 5 or 6 in regard to this some time back, and they were all over the place.

mirh, 2023-05-10:
Even just for the most stupid "incorrect login" or "wrong mountpoint" scenarios? Jeez. 
Well, at the most horrific possible worst you could just have a whitelist/lookup table for each different server software I guess..
(and no, I wished I had the time to code all of this, on top of barely being able to make comprehensive enough reports)

Though at the most simplest end instead, a timid start could just be separating the "everything is dead silent" case from the "any other problem" one.

MrPoke21, 2023-12-06:
[image]

mirh, 2023-12-07:
That's some nice start, but an error/message box could be appropriate if you are in *FormNtrip* trying to do the first setup. 
I certainly wouldn't block the whole screen just because the server was down.

closed 2024-02-15 by farmerbriantee

mirh, 2024-04-18:
This wasn't fixed, and not even the promising #348 was merged.