AgOpenGPS-Official/AgOpenGPS #340: AgIO GUI displays incorrect LAT and LON Data

chriskinal, 2023-11-06:
The OGI box on the System Data screen shows the correct LAT / LON data of the PAOGI sentence. However, the display of this data in upper left corner of the System Data screen and the main AgIO screen is wrong. The whole degrees are right but everything after the decimal point is wrong. It appears the correct data is sent to the AOG application as the "tractor" is in the right location. 
[image]

lansalot, 2023-11-06:
I think those are the raw, uncooked values. In the AOG code, you'll find the calculations used to convert them.

chriskinal, 2023-11-06:
Not sure what is meant by "raw, uncooked" values. The OGI box contains the correct GPS coordinates multiplied by 100 to move the decimal point two places right. The AgIO GUI seems to multiply them by 100 but butchers everything after the decimal point.

I believe the LAT display should show 32.3542691 not 32.5904485.

lansalot, 2023-11-06:
Found the code, it's here:
https://github.com/farmerbriantee/AgOpenGPS/blob/master/SourceCode/AgIO/Source/Forms/NMEA.Designer.cs

ParsePANDA and ParsePAOGI methods. It explains it there - it's the conversion from Degrees to Decimal Degrees I think.

[image]

[image]

chriskinal, 2023-11-06:
Got it.

chriskinal, 2023-11-06:
After seeing the conversion code the values are correct.

closed 2023-11-06 by chriskinal

mirh, 2023-11-07:
One could also argue it's a bit disturbing to have different latitude and longitude string orders depending on the screen you open.