AgOpenGPS-Official/AgOpenGPS #344: Lightbar adjustment missing?

DaylightPlains, 2023-11-24:
New to AOG, I suspect the ability to adjust the lightbar distance per pixel is missing in this version?

mirh, 2024-01-13:
Yup, that happened in 3a60fd7644f37d8512e5e3dfeee46e241d22eec6
Not sure how much it was meant, if voluntary

closed 2024-01-29 by richardklasens

kknuut, 2024-04-09:
This is needed feature for driving without autosteer. Currently bar is way too sensitive to be usable.

Request opening this and allowing at least manual adjustment via .config value if UI setting is too much for now.

Would only need reverting this  https://github.com/farmerbriantee/AgOpenGPS/commit/3a60fd7644f37d8512e5e3dfeee46e241d22eec6#diff-afa0a07266b4caab7e136e97f5b80e86c2c71a5e40c91388616f9be756a0b867

to here

https://github.com/farmerbriantee/AgOpenGPS/blob/dabc3b6ef6b19696af6d53919b43d556a1e89d31/SourceCode/GPS/Forms/GUI.Designer.cs#L30

https://github.com/farmerbriantee/AgOpenGPS/blob/dabc3b6ef6b19696af6d53919b43d556a1e89d31/SourceCode/GPS/Forms/GUI.Designer.cs#L674