AgOpenGPS-Official/AgOpenGPS #345: Use IMU as a WAS

markonovic00, 2023-11-26:
Hello,

For one of my subjects in college, we have to improve some open-source projects. 
As I am using AOG myself on our farm I would like to write a code for using IMU as an WAS.

The problem is that for the subject the PR shall be reviewed and merged afterwards. 
Would that be possible? 

Thank you in advance.

@farmerbriantee

richardklasens, 2023-11-26:
You can create a fork and make it, test it, and later create a PR?

mirh, 2023-11-26:
You can start by reading the previous episodes of the matter
https://discourse.agopengps.com/t/aog-without-wheel-angle-sensor/2929/73
https://github.com/v12cat/WAS-Less-AOG
https://github.com/jkonno/AOG_wasless

p.s. I think this would belong more to the [AgOpenGPS_Boards](https://github.com/farmerbriantee/AgOpenGPS_Boards) repo.

Davidwedel, 2023-11-26:
> The problem is that for the subject the PR shall be reviewed and merged afterwards.
> Would that be possible?

So the PR **HAS** to be merged? Is there any room for the dev team to say "we don't feel this is necessary right now"?
If you want my thoughts on WASless, make it an option to go WASless, not mandatory. I have tractors that I'd love to have 6 in accuracy for mowing with, but other tractors where if WAS is .5 in more accurate than WASless, I'll have a WAS.

mirh, 2023-11-27:
> So the PR HAS to be merged? Is there any room for the dev team to say "we don't feel this is necessary right now"?

I think the general way of thinking of the professors is that "of course if you have a good enough idea, any pedestrian project will accept it". But I would argue the dynamics here are very much different from the majority of the software projects. 

Merging isn't probably required strictly speaking, but since a random teacher cannot know what the hell every single program is about, they rely on the maintainers to rate the contribution.

markonovic00, 2023-11-27:
I confirmed with my professor that it is not required. 
So if you are okay with this being created as a new feature in the AgOpenGps_Boards repo I will gladly do it. And afterward, I will do the testing. 

To go to the point what my professor wanted was to contribute to some open-source project one way or another. 

I will need some code reviews if that will be possible. 
Thank you for the information. 

I will put a link to my forked repo with the changes when I am finished.

lansalot, 2023-11-27:
Go for it, sounds like a great project to me !

Pat-I, 2023-11-27:
@markonovic00 
It will be nice to see your ideas about how to integrate it.
So far all WAS less projects seemed to take place on the autosteer module, without modifications needed to AOG.

@mirh gave you the links to the most achieved project so far.

And this would need real field testing, not like a project that can run only on a computer.

A little off topic:
If your (or classmate) skills are good, there is my OpendGrade3D project that would benefit greatly from some help, especially for the graphical display(map display) or communication
Just saying if someone is still looking for a project ;)

mirh, 2023-11-28:
Another little off topic:
You may also look into integrating the [LSM9DS1](https://github.com/farmerbriantee/AgOpenGPS_Boards/issues/12) or ZED-F9R imus (assuming any of them can stand a chance compared to the BNO085), if you were looking for other "low level ideas" and not dirtying your hands with the main program.

richardklasens, 2024-01-29:
Will be moved to other repo

closed 2024-01-29 by richardklasens