AgOpenGPS-Official/AgOpenGPS #391: Curves parallel to each other or crossing cause automatic steering to get confused

GNSS-Stylist, 2024-04-23:
Fields with shapes that are not convex may be divided into two or more "sections" when using a boundary curve and an "inner" curve is selected. These sections are connected with two straight lines that are very close to each other. This may lead to automatic steering getting confused so that it doesn't seem to always know which one of those lines it should follow.

Made more thorough testing and videos with this external simulator: https://github.com/GNSS-Stylist/AgOpenGPSSimPoC but the same issue also occurs with the internal simulator and in a real tractor (tested also in a field to make sure this is not just a simulator glitch). Actually very likely stumbled upon this while driving last summer with AOG V5.7.2.

Video:

https://github.com/farmerbriantee/AgOpenGPS/assets/50438441/edd3c106-9930-425e-8f0c-8f6d94d88be9

Based on the lookahead-pointer (yellow square) jumping back and forth it seems that the nearest point is always selected from the whole curve (also by looking at the code it seems to do so)

This also occurs with AB-curve that has self-crossing parts (may not be the most sensible curve though...):

https://github.com/farmerbriantee/AgOpenGPS/assets/50438441/1d97062b-e6e3-46de-b378-3322d0379111

I already took a look at the code and got a fix to this somewhat working. It does, however still have some problems with some tight hairpins when a big look ahead-value is used and tractor is taking a shortcut. I will continue investigating, so assigning this to myself for now (not sure if I can do that, though. But will try at least...).

Field used in the videos above: 
[Field_Sim_Blob.zip](https://github.com/farmerbriantee/AgOpenGPS/files/15074113/Field_Sim_Blob.zip)

GNSS-Stylist, 2024-12-08:
This (now closed) PR doesn't seem to be linked to this automatically, so adding a link: https://github.com/farmerbriantee/AgOpenGPS/pull/392

richardklasens, 2024-12-09:
Pulled into its own branch

closed 2024-12-09 by richardklasens