AgOpenGPS-Official/AgOpenGPS #405: Window hidden because it appears out of screen

gunicsba, 2024-05-21:
There are ways to make the window go outside of the actual screen.
See video of bricbric: https://youtu.be/VL7Nh5VCQ70

Another easy to reproduce is when you use the tablet inhouse and put AOG on the external big screen. Take it to the tractor and it still starts on the external (currently not connected, not visible) screen.

@lansalot  did found a class that adress this:
https://learn.microsoft.com/en-us/dotnet/api/system.windows.forms.screen?view=windowsdesktop-8.0&redirectedfrom=MSDN

FarmerKevin, 2024-05-21:
I have experienced annoyance with the window shrinking, or moving if finger slides on screen (bouncing through field). Greatly helps if you steady hand with fingers on outside on tablet and use thumb to push. Trying to get others to do the same is the issue. Would b nice if it only changes with minimize, or buttons intended. Thanks! Keep up the great work!Kevin Alward

https://learn.microsoft.com/en-us/dotnet/api/system.windows.forms.screen?view=windowsdesktop-8.0&redirectedfrom=MSDN

farmerbriantee, 2024-05-31:
Resolved - puts any offscreen windows to 0,0

closed 2024-05-31 by farmerbriantee