AgOpenGPS-Official/AgOpenGPS #406: angle sensor

ogfdean, 2024-05-23:
Angle sensor has input and output but the system does not recognize it. not being tech savvy can i get a detailed description of how to solve this problem in plain language?

Davidwedel, 2024-05-24:
Hello. There's a forum for questions, that a lot more people watch than this. 
https://discourse.agopengps.com/

You might get farther asking on there.

ogfdean, 2024-05-24:
Thanks will review ag open gps more,  have tried their posted  solutions  to no avail. Getting to point that the provided board is faulty. Sent from Mail for Windows From: David WedelSent: Friday, May 24, 2024 6:47 AMTo: farmerbriantee/AgOpenGPSCc: ogfdean; AuthorSubject: Re: [farmerbriantee/AgOpenGPS] angle sensor (Issue #406) Hello. There's a forum for questions, that a lot more people watch than this.https://discourse.agopengps.com/You might get farther asking on there.

farmerbriantee, 2024-05-31:
This is not an issue with the software, but with hardware

closed 2024-05-31 by farmerbriantee

ogfdean, 2024-05-31:
Still looking for solutions. With a repair technician