AgOpenGPS-Official/AgOpenGPS #416: Pivot guidance path not constant radius circle

emcmar, 2024-10-09:
Cross posted not sure where issues are "officially" tracked.

I have been a user for a few years now. Mostly farm in irrigated circles. I have used ABcurve but wanting to transition to the pivot guidance track for various reasons. I have been testing it lately as it has never seemed to work as expected.

Upon testing I have found the generated path to not be a constant radius. I have tested this in simulator using flags and checking the distance to the pivot. I have also tested in the physical world with similar results with the delta in error increasing as the distance from the center increases. I have also had issues snapping the guidance line to the machines location but I think that can be explained by the guidance path being a non constant radius.

I have been digging through the code some but not found where the paths are built.

mirh, 2024-10-09:
Could it be related to #392?

emcmar, 2024-10-09:
Im not positive but it seems the issue is more in the calculations in how the circular path is built. I think this in part because the error seems relative to distance from the center point.

I have been using a version prior to your #392 changes and am excited to try out your changes in a AB curve scenario.

GNSS-Stylist, 2024-10-27:
I don't this is related to https://github.com/farmerbriantee/AgOpenGPS/pull/392 . That PR only affects how already constructed guidance line is being followed (doesn't allow switching to another line while already "locked" to some line).

closed 2024-11-18 by P00qwerty