AgOpenGPS-Official/AgOpenGPS #421: AOG fatal error when calculating the next 'turn' trajectory

simmoinard, 2024-10-31:
Using AOG V6.3.3 with AIO micro pcb on a Valtra N143 tractor.

Everything is working good, but in some cases when the turn is finished and it calculates the trajectory for the next turn, There is an error forcing AOG to close. The error is above, sorry it's in french but the main meaning is : "Index was out of scope. It must not be negative and must be less than the size of the collection."

Any idea ? Maybe I made an impossible way to turn according to the parameters I gave ? But in the case it would be better to just make an alarm, not a force-quit the software ?

```Consultez la fin de ce message pour plus de détails sur l'appel du débogage
juste-à-temps (JIT) à la place de cette boîte de dialogue.

************** Texte de l'exception **************
System.ArgumentOutOfRangeException: L'index était hors limites. Il ne doit pas être négatif et doit être inférieur à la taille de la collection.
Nom du paramètre : index
   à System.ThrowHelper.ThrowArgumentOutOfRangeException(ExceptionArgument argument, ExceptionResource resource)
   à AgOpenGPS.CYouTurn.CreateCurveOmegaTurn(Boolean isTurnLeft, vec3 pivotPos)
   à AgOpenGPS.CYouTurn.BuildCurveDubinsYouTurn(Boolean isTurnLeft, vec3 pivotPos)
   à AgOpenGPS.FormGPS.UpdateFixPosition()
   à AgOpenGPS.FormGPS.ReceiveFromAgIO(Byte[] data)
   à System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   à System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   à System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   à System.Windows.Forms.Control.InvokeMarshaledCallback(ThreadMethodEntry tme)
   à System.Windows.Forms.Control.InvokeMarshaledCallbacks()

************** Assemblys chargés **************
mscorlib
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9277.0 built by: NET481REL1LAST_B
    CodeBase : file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
AgOpenGPS
    Version de l'assembly : 6.3.3.0
    Version Win32 : 6.3.3
    CodeBase : file:///C:/Users/user/Downloads/AgOpenGPS_v63/AgOpenGPS.exe
----------------------------------------
System.Windows.Forms
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9251.0 built by: NET481REL1LAST_C
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9261.0 built by: NET481REL1LAST_C
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9037.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9037.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9277.0 built by: NET481REL1LAST_B
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
System.Xml
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9037.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
OpenTK.GLControl
    Version de l'assembly : 1.2.0.0
    Version Win32 : 1.2.2336.6514-pre
    CodeBase : file:///C:/Users/user/Downloads/AgOpenGPS_v63/OpenTK.GLControl.DLL
----------------------------------------
OpenTK
    Version de l'assembly : 1.2.0.0
    Version Win32 : 1.2.2336.6514-pre
    CodeBase : file:///C:/Users/user/Downloads/AgOpenGPS_v63/OpenTK.DLL
----------------------------------------
RepeatButtonControl
    Version de l'assembly : 1.0.0.0
    Version Win32 : 1.0.0.0
    CodeBase : file:///C:/Users/user/Downloads/AgOpenGPS_v63/RepeatButtonControl.DLL
----------------------------------------
mscorlib.resources
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9037.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_fr_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------
AgOpenGPS.resources
    Version de l'assembly : 6.3.3.0
    Version Win32 : 6.3.3
    CodeBase : file:///C:/Users/user/Downloads/AgOpenGPS_v63/AgOpenGPS.exe
----------------------------------------
AgOpenGPS.resources
    Version de l'assembly : 6.3.3.0
    Version Win32 : 6.3.3
    CodeBase : file:///C:/Users/user/Downloads/AgOpenGPS_v63/AgOpenGPS.exe
----------------------------------------
Microsoft.GeneratedCode
    Version de l'assembly : 1.0.0.0
    Version Win32 : 4.8.9037.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
----------------------------------------
System.Management
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9037.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll
----------------------------------------
Control.Draggable
    Version de l'assembly : 1.0.7235.37687
    Version Win32 : 1.0.7235.37687
    CodeBase : file:///C:/Users/user/Downloads/AgOpenGPS_v63/Control.Draggable.DLL
----------------------------------------
Accessibility
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9037.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/Accessibility/v4.0_4.0.0.0__b03f5f7f11d50a3a/Accessibility.dll
----------------------------------------
Keypad
    Version de l'assembly : 5.2.3.0
    Version Win32 : 5.2.10.0
    CodeBase : file:///C:/Users/user/Downloads/AgOpenGPS_v63/Keypad.DLL
----------------------------------------
System.Windows.Forms.MapControl
    Version de l'assembly : 1.1.0.0
    Version Win32 : 1.1.0.0
    CodeBase : file:///C:/Users/user/Downloads/AgOpenGPS_v63/System.Windows.Forms.MapControl.DLL
----------------------------------------
System.Windows.Forms.resources
    Version de l'assembly : 4.0.0.0
    Version Win32 : 4.8.9037.0 built by: NET481REL1
    CodeBase : file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_fr_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------

************** Débogage JIT **************
Pour activer le débogage juste-à-temps (JIT), le fichier de configuration pour cette
application ou cet ordinateur (machine.config) doit avoir la valeur
jitDebugging définie dans la section system.windows.forms.
L'application doit également être compilée avec le débogage
activé.

Par exemple :

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

Lorsque le débogage juste-à-temps est activé, les exceptions non gérées
seront envoyées au débogueur JIT inscrit sur l'ordinateur
plutôt que d'être gérées par cette boîte de dialogue.
```

richardklasens, 2024-12-10:
Please post a screenshot of your vehicle measurements, U-turn settings and implement setup.

richardklasens, 2024-12-10:
Is it happen in multiple fields?

closed 2024-12-14 by farmerbriantee