AgOpenGPS-Official/AgOpenGPS #570: Improve folder structure

FantasyTeddy, 2025-01-11:
The current folder structure of the repository looks like this:

- SourceCode
  - AgOpenGPS.sln
  - AgIO
    - Source
      - AgIO.csproj
  - AgLibrary
    - AgLibrary.csproj
  - GPS
    - AgOpenGPS.csproj
  - GPS_Out
    - Source
      - GPS_Out.csproj
  - Keypad
    - Keypad.csproj
  - ModSim
    - Source
      - ModSim.csproj

There are a few inconsistencies:
1. Some projects have an additional _Source_ folder.
2. The _AgOpenGPS_ project lives in a _GPS_ folder which does not match the project name.

I propose to remove the additional _Source_ folders and rename the _GPS_ folder to _AgOpenGPS_.

FantasyTeddy, 2025-01-12:
The only thing I am worried about with this, is that it will create a lot of potential merge conflicts for any unmerged changes.
I think everyone must agree to this and the timing must be right.

J005t67, 2025-01-13:
There are so many unmerged changes in the world, that the timing will never be right propbably.
I think my merge tooling can handle this. (I'm using Fork, and BeyondCompare for difficult cases), but we should not assume that other people have the same tooling and/or experience. So I have to reconsider my earlier comment: If it was a pull request, I would not approve it.

richardklasens, 2025-01-13:
I think it’s a good solution but maybe not for now

FantasyTeddy, 2025-01-13:
Okay. When would be a better point in time to do it?

richardklasens, 2025-01-13:
Maybe best to 

- wait for Joost his updates for a WFP demo
- Continue refactor on smaller and not drastic changes like this
- when the release is stable enough to continue on develop. It means there won’t be any other releases from 6.6.x until develop can poop out a stabilized pre-release.

FantasyTeddy, 2025-01-13:
Okay, but you do realize that:
> * wait for Joost his updates for a WFP demo

The changes required to achieve this will most likely result in much harder to deal with merge conflicts.

> * Continue refactor on smaller and not drastic changes like this

Okay, so you consider the work from Joost as "smaller"?

> * when the release is stable enough to continue on develop. It means there won’t be any other releases from 6.6.x until develop can poop out a stabilized pre-release.

This one I can understand and it would probably make sense to wait for the feedback from the release to cool down.

J005t67, 2025-01-13:
No need to  worry about timing in relation with my changes and certainly don't wait for me to finish something on AOG2. I will not merge them, but selectively copy them when they are ready. 
Although the current folder structure does not look good, I don't think it hurts in terms of maintainability, speed of development or preventing bugs. So, better not change it because the risc is big and the advantages minimal.

closed 2026-10-30 by richardklasens