AgOpenGPS-Official/AgOpenGPS #571: Remove unused code/resources

FantasyTeddy, 2025-01-11:
Because AgOpenGPS has a long history and has seen many changes and additions, there are quite a few things that are not used anymore.
In order to keep the codebase clean and maintainable, unused code or resources should be removed.

- [ ] Remove commented out code
- [ ] Remove code that is never called (dead code)
- [ ] Remove forms that are never displayed
- [ ] Remove resources (images, sounds, ...) that are not used anymore
- [ ] Remove settings that are not used anymore
- [ ] Remove texts/translations that are not used anymore
- [ ] Remove tools that are not used anymore (e.g. OGL.exe, UDP_SimTool.exe)

J005t67, 2025-01-11:
Commented out code sometimes contains usefull comments. I would keep the commented out code.

FantasyTeddy, 2025-01-12:
Can you give an example of where this is the case?
Because in my experience, commented out code has almost always been something that someone tried and then forgot about.

J005t67, 2025-01-13:
I agree that most of the commented out has no value anymore, but I propose to keep it in for now. And I don't want to dig in the code for hours to see which comments are still valuable and which ones are  not. I propose to remove commented out code l ater when somebody is working on a file anyway.

FantasyTeddy, 2025-01-13:
Okay, but this means that we also cannot do any of the other proposed cleanup steps: anything could be referenced from within commented out code and if we remove it, the commented out code might become useless...

Henny20, 2025-03-15:
To be honest. Images in Resources will not work on other platforms. There is no System.Drawing. That is only supported on Windows. We load images in C# or Xaml code.

richardklasens, 2025-03-16:
I filled the checkboxes :)

FantasyTeddy, 2025-03-16:
> I filled the checkboxes :)

What do the checked boxes represent for you? The things we should clean up?
Can you explain your reasoning for why these should be done and the others not?

richardklasens, 2025-03-16:
> What do the checked boxes represent for you? The things we should clean up? Can you explain your reasoning for why these should be done and the others not?

I think, based on my second reading, everything is ok but I’m with Joost, for understanding other parts it’s not a problem to leave commented code for now.

FantasyTeddy, 2025-03-16:
Okay, thank you for clarifying.
I guess we just need to be careful to not delete anything that is referenced from any commented out code.

FantasyTeddy, 2025-03-16:
I updated the description of this issue accordingly and removed the checkmarks again, such that they can be checked once they are actually done.

closed 2025-03-24 by FantasyTeddy