AgOpenGPS-Official/AgOpenGPS #579: Simplify code that uses pgn classes

J005t67, 2025-01-14:
I want to simplify the code that uses PGN classes

In the current situation, the calling code is responsible for:
1) scaling the value to make it fit in one or more bytes,
2) convert the scaled value to one byte ore more bytes,
3) put the bytes in the correct position in the message.

Just one of the many examples of the current code:

            p_239.pgn[p_239.speed] = unchecked((byte)(avgSpeed * 10));

When we shift these responsibilites to the pgn classes, the calling code would become a lot more simple.

            p_239.SetSpeedInKmh(avgSpeed_KMH);

Ofcourse the 'complex' code still exists, but it is more concentrated in the pgn classes and that makes it more easy debug and to spot bugs.
It is also more according to the design principle 'Separation of Concerns'

This change also enables more improvement that are not possible whith the current code that s scattered over many different files.
I want to replace all occurences of 'unchecked' with something more robust, that does check for any overflow problems. 'unchecked' basically sweeps the
overflow problems under the carpet.

My prefered way of working is:
1) I (or somebody else) change just 1 pgn class and its calling code and makes a PR, (without addressing the 'unchecked' issue.)
2) After the review we change all pgn classes, taking the review comments into account.
3) We complete this PR before starting on new topics, (like the 'unchecked')

FantasyTeddy, 2025-01-14:
I was actually planning to do something very similar (my bad for not creating an issue first).

It would probably be best if you make the first PR yourself, to given an example of how you are imagining the new PGN classes.

FantasyTeddy, 2025-01-14:
I was imagining something like this: https://github.com/FantasyTeddy/AgOpenGPS/commit/9930450d0ce55c293d812e8e18eecfca3c2c4a14

Let me know if this would match your expectations (and yes, better names than `PGN_DD` or `p_239` would certainly not hurt ;)

lansalot, 2025-01-14:
That looks ideal, especially with the better names.

J005t67, 2025-01-14:
Yes, a base class (or superclass) would be nice for all the reasons you mention. And also to avoid duplicate code for overflow checking and corresponding error logging (if any). But still, I want to start by just simplify the calling code and moving everything to the pgn classes.

Pynee, 2025-01-26:
How about if the "CRC" code was changed to a real CRC calculation for better error detection? As of now, it only sums up every byte in the message and that's not really robust way of detecting errors in a message.

FantasyTeddy, 2025-01-26:
> How about if the "CRC" code was changed to a real CRC calculation for better error detection? As of now, it only sums up every byte in the message and that's not really robust way of detecting errors in a message.

Unfortunately, this would break the current protocol and would require all the hardware and software to be updated to the new (better) CRC calculation (or support both for backwards compatibility).
This has already been discussed in #546 and I would recommend to continue any further discussion about that there.

This issue is mainly about the already open draft PR #585 and only refactors code, without changing the functionality.

Pynee, 2025-01-26:
Yeah sorry. I just thought it would be easier to implement the correct CRC calculation while the checksum is still ignored but you are right that this isn't the right thread to discuss it.

richardklasens, 2025-07-29:
also fixes #582

closed 2026-10-30 by richardklasens