AgOpenGPS-Official/AgOpenGPS #582: Incorrect CRC's in PGN 200 and 202

GwnDaan, 2025-01-14:
https://github.com/AgOpenGPS-Official/AgOpenGPS/blob/8903e66e6b0535edecf71ef63b48967c1b3c815a/SourceCode/AgIO/Source/Forms/FormUDP.cs#L136

The CRC is incorrect for this packet (pgn=202). It should be `0xE5`. 

https://github.com/AgOpenGPS-Official/AgOpenGPS/blob/8903e66e6b0535edecf71ef63b48967c1b3c815a/SourceCode/AgIO/Source/Forms/UDP.designer.cs#L76

Same for the CRC of the packet above (pgn=200). That should've been `0x82`. 

Probably this is some copy-paste mistake.

GwnDaan, 2025-01-14:
Same goes for pgn=201, that should be crc `0x4C`

J005t67, 2025-01-15:
I'm refactoring all Pgn related code, to make it more easy to use, and to avoid lots of duplicate code. All the new pgn classes use the same base class PgnBase. PgnBase computes the crc automatically when someone want to know the composed message, (so it is impossible to forget this).
Don't know when this will become available, but when it is available, the crc bugs are easily fixed by replacing the current code with a new Pgn class.