AgOpenGPS-Official/AgOpenGPS #597: Building AgOpenGPS not possible if downloaded instead of cloned

FantasyTeddy, 2025-01-19:
It is currently not possible to build the AgOpenGPS solution if the repository is downloaded ("Download ZIP") instead of cloned using Git.

> error MSB3073: The command ""C:\Users\user\.nuget\packages\gitversion.msbuild\5.12.0\tools\net48/gitversion.exe" "C:\Users\user\Downloads\AgOpenGPS-develop\AgOpenGPS-develop\SourceCode\AgLibrary"  -output file -outputfile obj\gitversion.json" exited with code 1.

Running the command on its own shows the reason for the failure:

> Cannot find the .git directory

In order to properly determine the version at build time, the `GitVersion.MSBuild` NuGet package requires access to the entire Git history, which is stored in the `.git` directory and only available when cloning via Git.

---

### Potential solutions

1. Ignore this issue and make it clear in the README, that the repository needs to be cloned and cannot be downloaded. (Already done in #584)
**Advantage**: Nothing needs to be changed.
**Disadvantage**: Everyone who still downloads the project instead of cloning it, will not be able to build it.
3. Remove the `GitVersion.MSBuild` NuGet package and invoke GitVersion only during the GitHub workflow.
**Advantage**: The project can be built even if it is downloaded instead of cloned.
**Disadvantage**: The version will only be calculated in the GitHub workflow, that means that when building locally, the version will need to be something static like `1.0.0` (or similar).

Let me know what you think, and if you see any other possible solutions.

lansalot, 2025-01-19:
In the build file, is there any way to conditionally include only if the .git folder is found?

richardklasens, 2025-01-19:
I would pick option to and set static to 0.9.9
Will it include the beta-line in the ogl?

FantasyTeddy, 2025-01-19:
> In the build file, is there any way to conditionally include only if the .git folder is found?

I think this is probably achievable, but I don't think that it is a good idea to complicate the build system too much...

> I would pick option to and set static to 0.9.9 Will it include the beta-line in the ogl?

Do you mean option 2?
Why would you pick 0.9.9 as the local development version?
Yes I think it should be pretty straight-forward to display something specific in this case.

richardklasens, 2025-01-19:
0.9.9 will result in something that it’s a build made by user and not official. Since it’s a rare number :)

richardklasens, 2025-01-23:
> [@AgOpenGPS-Official/aog-coders](https://github.com/orgs/AgOpenGPS-Official/teams/aog-coders): Are there any objections to implementing option 2?
> 
> The consequence would be that any local build (even if properly cloned via Git) will **not** display the "correct" version anymore.

I think it’s good to add, maybe put in a watermark with local-build with version 1.0.0.0 or something?

anuwon, 2025-01-23:
Visual Studio replication error
git failed with a fatal error. unable to access https://github.com/agopengps-official/agopengps/"1error setting certificate file."c/program files/git/mingw6-4/etc/ssl/certs/ca-bundle.crt 

Please provide a solution

closed 2025-01-24 by richardklasens