AgOpenGPS-Official/AgOpenGPS #598: Feature request: having a simplied mode in AOG to hide advanced functions

richardklasens, 2025-01-19:
Based on some users who setup a system and it will be runned by others or "dads" they like to have a simplified mode where a advanced features are disabled. What the advanced features are, is not determined yet.

FantasyTeddy, 2025-01-19:
In my opinion, this could also be solved by re-thinking the UI in general: moving the lesser used tools into sub-menus (or something similar).

Having a toggle switch that enables certain futures might not be ideal, because the features that should be enabled might vary from person to person. And soon we'll end up with multiple toggles for various features, which might be harder to maintain.

J005t67, 2025-01-20:
The features that one likes to include will vary from person to person, and also for each person from time to time. So be prepared that  people want every featuer to be configurable separately. Obviously, this will be a lot of work and require a lot of configuration items.
I expect that the biggest propblem will be in the conditional layout of the screen. The current WinForms implementation is really not prepared for that. Better wait until we have real progress on a new Ui implementation, that relies more on the layout capabilities of the controls like DockPanel, StackPanel, Grid etc. (In that case the screen layout will still look good if we conditionally, hide screen elements.)

richardklasens, 2025-01-20:
I understand completely.. this is just here to put the ideas from users that it wont be forgotten. 

For now I will close as not done

closed 2025-01-20 by richardklasens